ROBOTIS-GIT/OpenCM9.04 #13: Code hang: while (Serial2.read() != -1) ;

KurtE, 2017-12-18:
I have some arduino code that was working on a few different Arduino platforms (AVR, Teensy 3.x and LC, Edison) that is now hanging. 

I was trying to flush out any data in a Uart input buffer, by doing: 

    while (Serial2.read() != -1) ;

And the code hung.  Obviously I could do it a different way like: 

    while (Serial2.available()) Serial2.read();

But the original code should work as well.  I sort of was surprised that the Arduino install only included core header files and not the sources.  I am assuming this project is the sources. 

The UARTClass.cpp code simply returns stuff from drv...

    int UARTClass::read( void )
    {
      rx_cnt++;
      return drv_uart_read(_uart_num);
    }

and the drv_uart_read code does not check if the queue is empty, like peek does. 

    int drv_uart_read(uint8_t uart_num)
    {
        int ret = -1;
        int index;

        index = drv_uart_rx_buf_tail[uart_num];

        ret = drv_uart_rx_buf[uart_num][index];

        drv_uart_rx_buf_tail[uart_num] = (drv_uart_rx_buf_tail[uart_num] + 1) % DRV_UART_RX_BUF_LENGTH;

        return ret;
    }

KurtE, 2017-12-18:
I should mention this causes other parts of the Arduino Stream code to not work as well.

That is if you do something like:  Note: XBSerial is defined as Serial2

    XBeeSerial.print(F("+++")); 
    XBeeSerial.flush();
    XBeeSerial.setTimeout(20);  // give a little extra time
    if (XBeeSerial.readBytesUntil('\r', ab, 10) > 0) {

The call to readBytesUntil will not work properly as it loops calling timedRead
 
   int c = timedRead();

Which in Stream.cpp is defined as: 

    int Stream::timedRead()
    {
      int c;
      startMillis = millis();
      do {
        c = read();
        if (c >= 0) return c;
      } while(millis() - _startMillis < _timeout);
      return -1;     // -1 indicates timeout
    }

OpusK, 2017-12-20:
Hello, @KurtE 
Thank you for letting us know about this issue.

you're right.
In order to be compatible with existing Arduino codes, the following must be fulfilled.

> The first byte of incoming serial data available (or -1 if no data is available) - int.

But we were just returning the size of the data. That is, if there is no data, it returns 0.
We will fix this and we will test out what you have mentioned and release it as soon as possible.

OpusK, 2017-12-21:
@KurtE 

I have modified the code and confirmed that it works.
However, it may take some time to release.

If you need to use it right away, please copy the [lib_f103.a](https://github.com/ROBOTIS-GIT/OpenCM9.04/blob/develop/arduino/opencm_arduino/opencm9.04_release/variants/OpenCM904/lib_f103.a) file in the [develop branch](https://github.com/ROBOTIS-GIT/OpenCM9.04/tree/develop) to the location where the ArduinoBoard package is installed.

closed 2018-01-05 by chcbaram