ROBOTIS-GIT/OpenCM9.04 #14: Wire Library Problem

leoxbox, 2018-01-14:
Hi,

I am using the i2c port to connect some sensors, but in some case, it does not work right! 

After many investigations on the communication protocol and testing with different sensors, I find it out that it because the Wire library for cm9.04 is not using an actual hardware i2c and it's just a software i2c protocol implementation. So, in case of the i2c modules only support 400kHz frequency, the soft i2c communication is not working.
In this regard, I find the wire library from Maple Mini that it is using the actual i2c hardware. My suggestion is to use that lib for the OpenCm9.04 as well. It supports 400Khz communication also. :)

OpusK, 2018-01-15:
Hi,

OpenCM 9.04 already supports HW I2C (I2Cdev).
However, as you say, the Wire library uses Software I2C.

If you want to use HW I2C, please use [I2Cdev](https://github.com/ROBOTIS-GIT/OpenCM9.04/tree/master/arduino/opencm_arduino/opencm9.04/libraries/I2Cdev).
And we will review your suggestion internally.

Thank you :)

leoxbox, 2018-01-15:
Hi,

Yes, you are right. The I2Cdev is exactly using the hardware i2c. :) 
But that would be perfect if the general wire library (that is used in many available examples and general Arduino libraries) to be the same, so no need to change the other libraries source code to work with CM9.04 hardware I2C... 

Thanks. 👍

OpusK, 2018-01-16:
We agree with the case of I2C slaves that require fast speeds.
(Because we can not use the example wire or code right away)

As mentioned previously,
We need to discuss internally.

Thanks :)

closed 2018-01-16 by OpusK