ROBOTIS-GIT/OpenCM9.04 #15: [keil]: AX12+ does not move, although I can communicate with it.

ar-msb, 2018-02-04:
I am trying to program AX12+ with keil on stm32f407VG. 
Now, I can communicate with the motor by changing data in the registers, and I have tested changing the LED status and settings, but the motor does not move at all.
I am giving commands to the motor without any RX pin, that is to say, the data pin is connected to TX pin on my board ( we do not need any feedback from the motor.) with 12v,4.5A power supply.
Any suggestion about where the problem might be?

Thanks in advance.

OpusK, 2018-02-05:
Hello, @ar-msb 

You said: 
> I am giving commands to the motor without any RX pin, that is to say, the data pin is connected to TX pin on my board ( we do not need any feedback from the motor.) with 12v,4.5A power supply.

As you can see from [the manual of AX12+](http://emanual.robotis.com/docs/en/dxl/ax/ax-12a/#specifications), it does Half Duflex asynchronous communication with one line.

Therefore, regardless of whether feedback is needed, if both signals overlap on one data line, communication will not occur.
It is not known at this time whether this is the cause, but if you are only sending TX without using a S/W flow control, there may be a problem depending on the timing.

If you need some reference code, please refer to OpenCM9.04 Code or OpenCR.
[driver code](https://github.com/ROBOTIS-GIT/OpenCM9.04/blob/master/arduino/opencm_arduino/opencm9.04/variants/OpenCM904/hw/driver/drv_dxl.c#L49)
[DXL workbench library](https://github.com/ROBOTIS-GIT/OpenCM9.04/blob/master/arduino/opencm_arduino/opencm9.04/libraries/DynamixelSDK/src/dynamixel_sdk/port_handler_arduino.cpp#L184)

ar-msb, 2018-02-05:
@OpusK 
I bought one CM9.04 today, and now it works fine. This board really omits a lot of problems. I will 
make a communication between CM9.04 and STM32F407 to solve our problems.
Thanks for your help.

closed 2018-02-05 by ar-msb