ROBOTIS-GIT/OpenCM9.04 #17: [openCM IDE]: I can not communicate with OPENCM9.04

ar-msb, 2018-02-06:
Hi again,
I used opencm9.04 and it was fine, but from this afternoon I can not cammunicate with it. I can not even upload a blink program on it. on arduino IDE it says :
[image]
and on openCM IDE it says:
[image]
Can anyone help me please?

OpusK, 2018-02-07:
Hi, @ar-msb 

First of all, please check if you are using COM port in serial monitor, etc.
If COMPORT is free and this problem persists, try downloading again via [recovery mode](http://emanual.robotis.com/docs/en/parts/controller/opencm904/#recovery-mode).

ar-msb, 2018-02-07:
@OpusK 
Thanks. I used the recovery mode, and fortuantely, the board is fine now.

closed 2018-02-07 by ar-msb