ROBOTIS-GIT/OpenCM9.04 #19: [OpenCMIDE]:I can not get more than 30RPM from my motor.

ar-msb, 2018-02-14:
Hi.
We are controlling a furuta pendulum using AX12+, but we can not get any velocity more than 3.14 rad/s or 30 RPM which we really need. I was wondering if something has to be set to reach the maximum velocity. 
Any help would be appreciated. 
Thanks in advance.

OpusK, 2018-02-19:
Hi, @ar-msb 

I do not know what configuration you are running in which environment, so I give you a simple answer first.
As noted in the official manual for the AX12 +, the no load speed is up to 59 RPM (at 12V). So if you only meet those conditions, you will get the speed you want.

First of all, please check whether 12V rated voltage is used and how much current is available.
Of course, please check whether the PWM is 100% duty ratio.

ar-msb, 2018-02-21:
@OpusK 
Thanks, for your response. I did not find any information about PWM in the manual. Do you mean torque limit?

OpusK, 2018-02-22:
No, what I mean is "No Load Speed".
You can find it by searching this word in the manual.

In addition, please refer to the [wheel mode](http://emanual.robotis.com/docs/en/dxl/ax/ax-12a/#moving-speed-32).

OpusK, 2018-03-09:
This issue will be closed since there was no actions for a while.
You can reopen this issue to show this issue to the users whenever.

closed 2018-03-09 by OpusK