ROBOTIS-GIT/OpenCM9.04 #20: Can not program OpenCM 9.04 using Arduino IDE...

KurtE, 2018-03-27:
This has been mentioned in a few different threads on Robotis forum, such as: http://en.robotis.com/service/forum_view.php?bbs_no=2406740&slg=&page_type=&mode=&submode=

I have an OpenCM9.04B board I purchased awhile ago probably from Trossen Robotics.  I can currently program this board using the Robotis_OpenCM IDE, however I can not program it using your Arduino board install as it fails.  It  also fails to use the failsafe to load a program or update the firmware using your different apps such as R+ Manager or Roboplus Manager. 

However I have some newer 904A and 904C boards which this same set works just fine.  

Not sure if some firmware changed at some point?  If so can I somehow reflash the 904B?  I do have an ST-Link which I have used in the past on the Trossen Arbotix-Pro.  

If not easily fixable, not a big deal, as these boards are pretty cheap.

KurtE, 2018-04-01:
Up on the forum, it was suggested in the thread: http://en.robotis.com/service/forum_view.php?bbs_no=2406739&slg=&page_type=&mode=&submode=
That the bootloader has not been updated... But I was curious, so wrote simple sketch to dump the first part of the bootloader as well as the 96 bit unique serial number.

```
#define LED_PIN 14
void setup() {
  pinMode(LED_PIN, OUTPUT);
  for (int i = 0; i < 10; i++) {
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
  while (!Serial && (millis() < 10000)) ; // wait up to 10 seconds for serial port
  Serial.begin(115200);
  delay(250);
  Serial.print("Start Memory dump SN:");
  Serial.print(*(uint32_t*)DEVICE_ID1, HEX);
  Serial.print(":");
  Serial.print(*(uint32_t*)DEVICE_ID2, HEX);
  Serial.print(":");
  Serial.println(*(uint32_t*)DEVICE_ID3, HEX);
  uint8_t *addr = (uint8_t*)0x8000000l;

  while (addr < (uint8_t*)(0x8000000l + 0x1000)) { // Maybe look at first 4K
    uint8_t b = *addr;
    if (b < 0x10) Serial.print("0");
    Serial.print(b, HEX);
    Serial.print(" ");
    addr++;
    if (((uint32_t)addr & 0xf) == 0) Serial.println();
  }
}

void loop() {
}
```
I ran it on two different recent 904A boards and the output matched except for the serial numbers...
The data started off like:
```
Start Memory dump SN:66DFF48:48518772:67095020
00 50 00 20 69 1F 00 08 31 11 00 08 33 11 00 08 
35 11 00 08 37 11 00 08 39 11 00 08 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 3D 11 00 08 
3B 11 00 08 00 00 00 00 3F 11 00 08 41 11 00 08 
55 11 00 08 57 11 00 08 59 11 00 08 5B 11 00 08 
5D 11 00 08 5F 11 00 08 61 11 00 08 63 11 00 08 
65 11 00 08 67 11 00 08 69 11 00 08 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 6B 11 00 08 6D 11 00 08 
```

Note: R+Manager shows the bootloader as version 1.03 
As I could not program the old board using Arduino IDE, I made a version for OpenCM IDe 
```
#define LED_PIN 14
#define         DEVICE_ID1          (0x1FFFF7E8)
#define         DEVICE_ID2          (0x1FFFF7EC)
#define         DEVICE_ID3          (0x1FFFF7F0)
#include <stdint.h>
void setup() {
  pinMode(LED_PIN, OUTPUT);
  for (int i = 0; i < 10; i++) {
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(100);
  }
  delay(5000);
  //SerialUSB.begin(115200);
  delay(250);
  SerialUSB.print("Start Memory dump SN:");
  SerialUSB.print(*(uint32_t*)DEVICE_ID1, HEX);
  SerialUSB.print(":");
  SerialUSB.print(*(uint32_t*)DEVICE_ID2, HEX);
  SerialUSB.print(":");
  SerialUSB.println(*(uint32_t*)DEVICE_ID3, HEX);
  uint8_t *addr = (uint8_t*)0x8000000l;

  while (addr < (uint8_t*)(0x8000000l + 0x1000)) { // Maybe look at first 4K
    uint8_t b = *addr;
    if (b < 0x10) SerialUSB.print("0");
    SerialUSB.print(b, HEX);
    SerialUSB.print(" ");
    addr++;
    if (((uint32_t)addr & 0xf) == 0) SerialUSB.println();
  }
  
  
}

void loop() {
  // put your main code here, to run repeatedly:

}
```
I ran the program and the output was very different:
```
Start Memory dump SN:671FF48:55526750:87243805
00 50 00 20 A9 19 00 08 9D 0A 00 08 9F 0A 00 08 
A1 0A 00 08 A3 0A 00 08 A5 0A 00 08 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 A9 0A 00 08 
A7 0A 00 08 00 00 00 00 AB 0A 00 08 AD 0A 00 08 
C1 0A 00 08 C3 0A 00 08 C5 0A 00 08 C7 0A 00 08 
C9 0A 00 08 CB 0A 00 08 CD 0A 00 08 CF 0A 00 08 
D1 0A 00 08 D3 0A 00 08 D5 0A 00 08 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 D7 0A 00 08 D9 0A 00 08 
```
So still wondering if there is a standard app or sketch that we can use to update older boards to the current firmware?

Note: I have seen a sketch to update to MAPLE bootloader: https://github.com/Gregwar/maple-bootloader-robotis/blob/master/sketch/opencm904_maple_loader/opencm904_maple_loader.ino

Also might be nice to look at sources of firmware to see how hard it would be to have the bootloaders USB descriptor output the iSerial ...

ROBOTIS-Will, 2018-04-02:
@KurtE 
Thank you for attaching the test result. It is very interesting because I've been seeing all OpenCM9.04 series have the exact same bootloader, but I might be wrong.
I was able to find the bootloader update record in the past so I'm going to check whether this update is relevant to Arduino IDE support.
It'll take a couple days to track the changes so please sit back and relax.

ROBOTIS-Will, 2018-04-02:
@KurtE 
The change in bootloader should not affect to Arduino IDE support feature as it only increases timeout margin from 200ms to 999ms.
Our engineer has gone through bootloader binaries and was not able to find the matching data with your "very different" one.
Can you identify your OpenCM9.04 bootloader version with R+Manager as shown in the manual?
[http://emanual.robotis.com/docs/en/edu/mini/#firmware-update](http://emanual.robotis.com/docs/en/edu/mini/#firmware-update)

KurtE, 2018-04-02:
@ROBOTIS-Will 

As I believe I have mentioned, none of the utility programs like the R+ Manager can identify the bootloader... They say bootloader not found... 

So for more information, I updated the test program to guess where the end of the bootloader was and I dump the whole thing.  I also print Ascii data which hopefully helps identify.

Here is the openCM IDE version:
```
#define         DEVICE_ID1          (0x1FFFF7E8)
#define         DEVICE_ID2          (0x1FFFF7EC)
#define         DEVICE_ID3          (0x1FFFF7F0)
#include <stdint.h>
#define DBGSerial SerialUSB
uint8_t *bootloader_end = (uint8_t*)0x8002fff;
uint8_t *addr = (uint8_t*)0x8000000;
void setup() {
  pinMode(BOARD_LED_PIN, OUTPUT);
  pinMode(BOARD_BUTTON_PIN, INPUT_PULLDOWN);
  for (int i = 0; i < 10; i++) {
    digitalWrite(BOARD_LED_PIN, !digitalRead(BOARD_LED_PIN));
    delay(100);
  }
  //while (!DBGSerial && (millis() < 10000)) ; // wait up to 10 seconds for DBGSerial port
//  DBGSerial.begin(115200);
  delay(250);
  DBGSerial.print("Try to find end of Bootloader last char: ");
  uint8_t last_char = *bootloader_end;
  DBGSerial.print(last_char, HEX);
  DBGSerial.print(") size:");

  while (*bootloader_end == last_char) {
    bootloader_end--;
  }
  int bootloader_size = (int)(bootloader_end - addr) + 1;
  DBGSerial.println(bootloader_size, DEC);
  
}

void loop() {
  char ascii_data[17] = "                ";
  addr = (uint8_t*)0x8000000;
  DBGSerial.println("Press the User button on board");
  while (!digitalRead(BOARD_BUTTON_PIN)) ;

  DBGSerial.print("Start Memory dump SN:");
  DBGSerial.print(*(uint32_t*)DEVICE_ID1, HEX);
  DBGSerial.print(":");
  DBGSerial.print(*(uint32_t*)DEVICE_ID2, HEX);
  DBGSerial.print(":");
  DBGSerial.println(*(uint32_t*)DEVICE_ID3, HEX);

  
  while (addr <= bootloader_end) { // Maybe look at first 4K
    uint8_t b = *addr;
    if (b < 0x10) DBGSerial.print("0");
    DBGSerial.print(b, HEX);
    DBGSerial.print(" ");
    ascii_data[((uint32_t)addr & 0xf)] = ((b >= ' ') && (b <= '~'))? (char)b : '.';
    addr++;
    if (((uint32_t)addr & 0xf) == 0) {
      DBGSerial.print("- ");
      DBGSerial.println(ascii_data);
    }
  }

}
```
I have included the completed outputs:  
There are some strings near the end of each of them... The 901A version (newer) shows the 1.03 in the string.  Did not find like string in the older version

[CM901A-1.txt](https://github.com/ROBOTIS-GIT/OpenCM9.04/files/1868234/CM901A-1.txt)
[CM901B.txt](https://github.com/ROBOTIS-GIT/OpenCM9.04/files/1868235/CM901B.txt)

ROBOTIS-Will, 2018-04-03:
@KurtE 
We were trying to find the match of CM901B.txt binary code, but we couldn't.
It could be a beta tester module that was not sold in the shop because it does not show the bootloader version.
At this point I have no clue how the unidentified bootloader is loaded in your old OpenCM9.04.

KurtE, 2018-04-03:
@ROBOTIS-Will 
Not sure, I ordered this board from Trossen Robotics back in March 2014.  I did not use it for a long time, and only recently tried it with the Arduino IDE. 

Would be interesting to see if anyone else has these 4 year old boards.   

I will ask up on Trossen Robotics Forum to see if anyone there might check what Bootloader they have.

ROBOTIS-Will, 2018-04-03:
@KurtE 
We digged a bit more and found that the earlier OpenCM9.04 before releasing ROBOTIS MINI(2014 1Q) was using bootloader for embedded only which means it does not support R+Manager, but OpenCM IDE.
After releasing ROBOTIS MINI, OpenCM9.04 bootloader had been updated for R+Manager.
So what you have now is seems to be the very early version of OpenCM9.04.
To me, it sounds like an ancient legacy xD
Thank you for helping us to find out the history of OpenCM9.04!

KurtE, 2018-04-03:
@ROBOTIS-Will 

Thanks, good to know.

My guess it is easier to just replace this board with newer one (I do have 3 new ones), but liked the old B as it had two AX Servo connectors...  

I wonder if it would make sense to try running a modified version of the Maple program updater, where maybe I simply use the bytes I could extract from one of the new boards...

My guess is a probably, wont fix...

KurtE, 2018-04-03:
@ROBOTIS-Will 
:) - The modified version of the Maple firmware update program mentioned earlier appears to work.  I compiled it and downloaded it using opencm IDE,  I then was able to program it with Arduino and the R+Manager appears to be happy with it.
[opencm904_bootloader_update-180403a.zip](https://github.com/ROBOTIS-GIT/OpenCM9.04/files/1872305/opencm904_bootloader_update-180403a.zip)

ROBOTIS-Will, 2018-04-04:
Oh, wow! Did you make it work?
That's awesome! I'll refer to this issue thread whoever is experiencing the same problem.
I appreciate your contribution.

closed 2018-04-04 by ROBOTIS-Will

Ash-S, 2020-02-19:
Hello @KurtE , I am facing a bit similar issue, please help me to resolve it. I want to upload compiled .bin to openCM904 using flash loader but failed. The boot code flash well but how can i flash the compiled bin like led_blink.bin for example. Please suggest! Thanks.

KurtE, 2020-03-03:
@Ash-S Sorry,  I did not see this one until now. 

My particular case was that I had an older version of the board, which Arduino would not program. 
I was able to use the Maple ... stuff as mentioned earlier to reprogram the board with current firmware and at the time board would work.

Unclear if you have the same problem or not.  Also since it has been almost 2 years now, I forgot most of the details.  Good luck

ROBOTIS-Will, 2020-03-13:
@KurtE Thank you Kurt. I believe Ash-S’s issue had been resolved from #94