ROBOTIS-GIT/OpenCM9.04 #23: xl430 not found wth roboplus wizard or dynamixel_workbench_single_monitor

LasseBoerresen, 2018-04-26:
HI, 

I can connect and control all my 3 servos using opencm904 + 485 using arduino IDE and the dynamixel_workbench s_monitor example script. But the dxl_tosser example script of the opencm IDE does not work with the xl430 motors. It works with my xl320. Using those, I can also use roboplus wizard. But roboplus wizard does not find the xl430 servos. What is the difference? Both servo types use protocol 2.0. I read that opencm IDE is being discontiued, and arduino IDE should be used. But I cannot find something similar to dxl_tosser in the arduino IDE examples. I want to use the tosser to have the same behaviour as USB2Dynamixel. How can I do that?

Lasse Børresen

OpusK, 2018-04-27:
@Hi, @LasseBoerresen 

Is the roboplus wizard you used [R+ Manager](http://emanual.robotis.com/docs/en/software/rplus2/manager/)? Or is it the [Dynamixel Wizard](http://emanual.robotis.com/docs/en/software/rplus1/dynamixel_wizard/)?

If it is Dynamixel Wizard, please try search using R+ Manager.
First of all, it is possible to move to the next step by checking whether it is search.

LasseBoerresen, 2018-04-28:
Hi, I was using R+1 Wizard. Now I tried R+2 Manager, and here I could recover firmware on my 9.04 to version 18, and now this board could be found by the manager. However, the manager still can't find the xl430. I tried recovering firmware on the xl430, through 9.05+485 board, but the firmware recovery failed, and now my xl430 does not even blink red when turning on. How can I recover my servo? Is it impossible with 9.04+485? What board should I use instead?

OpusK, 2018-04-30:
Using the R+2 Manager, will not xl430 be found when I try all baud rates?
And, Is it not found when connecting only the xl430 using USB2Dynamixel?

And the recommended voltage of the [xl430](http://emanual.robotis.com/docs/en/dxl/x/xl430-w250/#specifications) is 11.1V. Are you supplying 11.1V power? (On the other hand, [xl320](http://emanual.robotis.com/docs/en/dxl/x/xl320/#specifications) is 7.4V.)

LasseBoerresen, 2018-04-30:
I am using 7.4V, but xl430 is stated to have operating voltage from 6.5 to 12. And it works fine when using opencm9.04 with native code.  I cannot find it on any baud rate. 
I do not have a USB2Dynamixel. Can I buy and use a u2d2 instead?

OpusK, 2018-04-30:
@LasseBoerresen.

Oh, I misunderstood your question and I was putting up an answer. Sorry.

> Can I buy and use a u2d2 instead?

Yes, you can buy and use a U2D2 at [shop](http://www.robotis-shop-en.com/?act=shop_en.goods_view&GS=3288&keyword=U2D2).
We recommend using U2D2 for more stable control.

> but the firmware recovery failed

If the firmware recovery of the xl430 fails, the xl430 will not function normally. Is that right?
I would like to know if the xl430 is still controllable through OpenCM9.04 DXL examples.

The function to recover Dynamixel F/W via R+2 Manager and OpenCM 9.04 will be released in 2 weeks.
This release is for the R+2 Manager.
For more information, please check for release on future.

> I want to use the tosser to have the same behaviour as USB2Dynamixel. How can I do that?

The Arduino example, which functions like tosser, will be added in a future release. There are some problems with USB transmission and reception, so we recommend that you do not use tosser. (The faster the speed, the more trouble occurs)

LasseBoerresen, 2018-05-03:
I will wait for the R+2 Manger update then. After the failed firmwarerecovery, the xl430 does not respond.

OpusK, 2018-05-04:
I will let you know when it's released :)

closed 2018-05-11 by robotpilot