ROBOTIS-GIT/OpenCM9.04 #24: Serial USB not displaying negative dynamixel position values

8035mpg, 2018-04-29:
I have an OpenCM9.04 connected to a MX-64AT dynamixel. I have set the dynamixel to Multi-Turn Mode. I command the dynamixel to move to different goal positions and use the "readWord" function to read the present position (addresses 36 and 37) and print to the Serial Monitor via USB. According to the e-manual for MX-64AT, in multi-turn mode, the dynamixel should be able to go to position values in the range -28672 to +28672. When the dynamixel is in positive position values, the position prints fine. But when it goes to negative values, all I get is "65535". Here's the code used to print out position:

```
//move to new goal within while loop
Dxl.writeWord(1, 30, newGoal);
while ( abs(newGoal - Dxl.readWord(1, 36)) > 10 )
{
      SerialUSB.print("Position: ");
      SerialUSB.print(Dxl.readWord(1, 36));
      SerialUSB.print(", Goal: ");
      SerialUSB.println(newGoal);
}
```

An excerpt of the output displayed on Serial Monitor as the dynamixel moves from positive position values to negative position values is as follows:

> Position: 1837, Goal: -7000
> Position: 1716, Goal: -7000
> Position: 1599, Goal: -7000
> Position: 1482, Goal: -7000
> Position: 1364, Goal: -7000
> Position: 1247, Goal: -7000
> Position: 1125, Goal: -7000
> Position: 1007, Goal: -7000
> Position: 890, Goal: -7000
> Position: 777, Goal: -7000
> Position: 664, Goal: -7000
> Position: 553, Goal: -7000
> Position: 439, Goal: -7000
> Position: 327, Goal: -7000
> Position: 214, Goal: -7000
> Position: 99, Goal: -7000
> Position: 65535, Goal: -7000
> Position: 65535, Goal: -7000
> Position: 65535, Goal: -7000
> Position: 65535, Goal: -7000
> Position: 65535, Goal: -7000
> Position: 65535, Goal: -7000
> Position: 65535, Goal: -7000
> Position: 65535, Goal: -7000
> Position: 65535, Goal: -7000
> Position: 65535, Goal: -7000
> Position: 65535, Goal: -7000
> Position: 65535, Goal: -7000
> Position: 65535, Goal: -7000
> Position: 65535, Goal: -7000
> Position: 65535, Goal: -7000
> Position: 65535, Goal: -7000
> Position: 65535, Goal: -7000
> Position: 65535, Goal: -7000
> Position: 65535, Goal: -7000

What am I doing wrong?

8035mpg, 2018-04-30:
Also, if 'Dxl.readWord()' is not recommended, what can I use instead?

OpusK, 2018-04-30:
You can refer to [e-Manual](http://emanual.robotis.com/docs/en/parts/controller/opencm904/#apis) for new APIs.

8035mpg, 2018-05-02:
Ok, I changed over to Arduino IDE and it is now able to read negative positions. But two questions about the new API.
(1) I noticed in the examples that they always write this at the beginning:
```
//setup comm with OpenCM9.04 board
#if defined(__OPENCM904__)
  #define DEVICE_NAME "1" //Dynamixel on Serial1(USART1)  <-OpenCM9.04
#elif define(__OPENCR__)
  #define DEVICE_NAME ""
#endif
```

This is not necessary, right? I am assuming I don't have to do this and can just write:
```
#define SERIAL1 1

dxl_wb.begin(SERIAL1, 57600);
```

(2) What is the purpose of this line of code? Is it even necessary?:
```
dxl_wb.ping(ID);
```

------------------------------
For reference, here is an excerpt of the output when moving from positive position values to negative position values after I got it to work with Arduino IDE and APIs:

> Position: 1171, Goal: -7000
> Position: 1101, Goal: -7000
> Position: 1027, Goal: -7000
> Position: 960, Goal: -7000
> Position: 891, Goal: -7000
> Position: 820, Goal: -7000
> Position: 750, Goal: -7000
> Position: 675, Goal: -7000
> Position: 608, Goal: -7000
> Position: 539, Goal: -7000
> Position: 466, Goal: -7000
> Position: 396, Goal: -7000
> Position: 323, Goal: -7000
> Position: 256, Goal: -7000
> Position: 186, Goal: -7000
> Position: 115, Goal: -7000
> Position: 45, Goal: -7000
> Position: -27, Goal: -7000
> Position: -95, Goal: -7000
> Position: -163, Goal: -7000
> Position: -234, Goal: -7000
> Position: -303, Goal: -7000
> Position: -377, Goal: -7000
> Position: -444, Goal: -7000
> Position: -514, Goal: -7000
> Position: -585, Goal: -7000
> Position: -655, Goal: -7000
> Position: -729, Goal: -7000
> Position: -796, Goal: -7000
> Position: -866, Goal: -7000
> Position: -938, Goal: -7000

And here are the parts of the code that set the dynamixel mode and print out the position and goal values:

```
  //set to multi-turn mode
  dxl_wb.itemWrite(ID, "CW_Angle_Limit", 4095); //CW limit
  dxl_wb.itemWrite(ID, "CCW_Angle_Limit", 4095); //CCW limit
  dxl_wb.itemWrite(ID, "Moving_Speed", 500); //set speed
```

```
  //move to new goal within while loop
  dxl_wb.itemWrite(ID, "Goal_Position", goal);
  while ( abs(goal - dxl_wb.itemRead(ID, "Present_Position")) > 50 )
  {
    Serial.print("Position: ");
    Serial.print(dxl_wb.itemRead(ID, "Present_Position"));
    Serial.print(", Goal: ");
    Serial.println(dxl_wb.itemRead(ID, "Goal_Position"));
  }
```

8035mpg, 2018-05-03:
How do I give a STOP command when the Dynamixel is turning in multi-turn mode (before it reaches the goal position)? I cannot set the speed to '0' because '0'-speed means maximum speed in multi-turn mode.

OpusK, 2018-05-04:
If you want to stop at a certain position, you should change "Goal Position" to that value.
Therefore, you should monitor the current position value and set the "Goal Position" to the desired value or the position value indicated by the current position.

8035mpg, 2018-05-08:
Okay, that worked. My emergency stop command in multi-turn mode is simply:

```
//give stop command in multi-turn mode
dxl_wb.itemWrite(ID, "Goal_Position", dxl_wb.itemRead(ID, "Present_Position"));
```

Thanks again for your help. I think you can go ahead and close this issue.