ROBOTIS-GIT/OpenCM9.04 #25: Incompatibility with Adafruit Graphics Libraries

brycebeddard, 2018-07-19:
Hello!

I am currently trying to use an Adafruit LCD display with the OpenCM9.04. It appears the libraries [GFX](https://github.com/adafruit/Adafruit-GFX-Library) and [ST7735] (https://github.com/adafruit/Adafruit-ST7735-Library) are not compatible currently with the board. I've attempted to modify the #include statements to no avail. Is there a certain library I should include that I'm just overlooking?

`Arduino: 1.8.5 (Windows Store 1.8.10.0) (Windows 10), Board: "OpenCM9.04 Board, OpenCM Bootloader"

C:\Users\intern-04\Documents\Arduino\libraries\Adafruit_GFX_Library\Adafruit_SPITFT.cpp: In constructor 'Adafruit_SPITFT::Adafruit_SPITFT(uint16_t, uint16_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t)':

C:\Users\intern-04\Documents\Arduino\libraries\Adafruit_GFX_Library\Adafruit_SPITFT.cpp:90:66: error: 'digitalPinToPort' was not declared in this scope

     dcport      = (RwReg *)portOutputRegister(digitalPinToPort(dc));

                                                                  ^

C:\Users\intern-04\Documents\Arduino\libraries\Adafruit_GFX_Library\Adafruit_SPITFT.cpp:90:67: error: 'portOutputRegister' was not declared in this scope

     dcport      = (RwReg *)portOutputRegister(digitalPinToPort(dc));

                                                                   ^

C:\Users\intern-04\Documents\Arduino\libraries\Adafruit_GFX_Library\Adafruit_SPITFT.cpp:91:41: error: 'digitalPinToBitMask' was not declared in this scope

     dcpinmask   = digitalPinToBitMask(dc);

                                         ^

C:\Users\intern-04\Documents\Arduino\libraries\Adafruit_GFX_Library\Adafruit_SPITFT.cpp:97:72: error: 'portInputRegister' was not declared in this scope

         misoport    = (RwReg *)portInputRegister(digitalPinToPort(miso));

                                                                        ^

C:\Users\intern-04\Documents\Arduino\libraries\Adafruit_GFX_Library\Adafruit_SPITFT.cpp: In constructor 'Adafruit_SPITFT::Adafruit_SPITFT(uint16_t, uint16_t, int8_t, int8_t, int8_t)':

C:\Users\intern-04\Documents\Arduino\libraries\Adafruit_GFX_Library\Adafruit_SPITFT.cpp:144:66: error: 'digitalPinToPort' was not declared in this scope

     dcport      = (RwReg *)portOutputRegister(digitalPinToPort(dc));

                                                                  ^

C:\Users\intern-04\Documents\Arduino\libraries\Adafruit_GFX_Library\Adafruit_SPITFT.cpp:144:67: error: 'portOutputRegister' was not declared in this scope

     dcport      = (RwReg *)portOutputRegister(digitalPinToPort(dc));

                                                                   ^

C:\Users\intern-04\Documents\Arduino\libraries\Adafruit_GFX_Library\Adafruit_SPITFT.cpp:145:41: error: 'digitalPinToBitMask' was not declared in this scope

     dcpinmask   = digitalPinToBitMask(dc);

                                         ^

exit status 1
Error compiling for board OpenCM9.04 Board.
`

OpusK, 2018-07-20:
Hi, @brycebeddard 

Thanks for your report.

First, I apologize for any inconvenience you may have experienced. Currently, OpenCM 9.04 does not implement the macro you mentioned. I think I need to implement this to fully support Arduino, and I will include it in a future release.

If you want to do it yourself, please refer to [this link](https://github.com/PaulStoffregen/cores/blob/master/teensy3/pins_arduino.h#L198).

brycebeddard, 2018-07-20:
Hi @OpusK 

Thanks so much for the quick response. The link did allow the example Graphics Test to compile; however, uploading it to the OpenCM causes the board to not respond and require a firmware recovery. Trying a few different things to see if there's a way to remedy this. Do you believe there is or have any ideas why the board would be reacting this way?

OpusK, 2018-07-23:
Did you just use those macros? Because the macros directly access register addresses, you must port them for OpenCM 9.04.
We are currently deferring this task by another task, but we also need to check if it can be ported so that it can be used as it is.

brycebeddard, 2018-07-23:
That explains it... Showing my ignorance. Going to see what I can do about that from the documentation available. Thanks!

OpusK, 2018-07-24:
In addition, I leave [a referenceable link](https://github.com/stm32duino/Arduino_Core_STM32/blob/master/cores/arduino/pins_arduino.h).

Since OpenCM also uses STM32, it seems to be possible, but it seems that it will take time to add and check many things.

KurtE, 2018-07-26:
Update: I received github email from Adafruit that they would accept a PR for this, so I went ahead and cloned gfx, made change and issued PR:
https://github.com/adafruit/Adafruit-GFX-Library/pull/168

OpusK, 2018-07-26:
@KurtE,

Thank you for your PR and contribution!
We'll still add this macro to our milestones and review it. Thanks again for your contribution.

KurtE, 2018-07-27:
@OpusK - 

I thought about trying to add the macros, but, the problems that I saw was that most of the places in libraries like this that use these macros, than do something like:
```
#define SPI_DC_HIGH()           *dcport |=  dcpinmask
#define SPI_DC_LOW()            *dcport &= ~dcpinmask
#define SPI_CS_HIGH()           *csport |=  cspinmask
#define SPI_CS_LOW()            *csport &= ~cspinmask
```
These macros in theory would work if the macros returned
GPIOX->ODR
And the other one returned the mask.   Than these macros in theory could work, but these operations would not be atomic and could lead to strange behaviour.  I have see this on other platforms, that is
the |= cspinmask, logically breaks into: 
Read the register
OR in the bit
write the register.

What happens if after the register is read, an interrupt happens and during that interrupt one (or more) of the states of other IO pins on that port changed, than or OR in/Write would then overwrite the updated values... 

The digitalWrite functions, instead use the BSRR, where you set a specific bit if you want the IO pin to be set HIGH and a different bit if you want it set low.  

On some of the Teensy boards, they handle this by using bit banding, where each of these bits are bit banded to a specific memory address, where you can set it high or low atomically. 

Section 3.3.2 of the STM32... manual looks like this board may support this.  Wonder if you could use this for that...

OpusK, 2018-07-27:
@KurtE -

In addition to Teensy, the STM32 Arduino core uses a register method.
As you say, it seems possible.

KurtE, 2018-07-27:
@OpusK ,

The bitbanding looks like a standard ARM M3, so was able to get the bitband address for the LED pin and blink it... 

Example program:
```
int led_pin = 14;
#define GPIO_BITBAND_ADDR(reg, bit) (((uint32_t)&(reg) - 0x40000000) * 32 + (bit) * 4 + 0x42000000)
#define GPIO_BITBAND_PTR(reg, bit) ((uint32_t *)GPIO_BITBAND_ADDR((reg), (bit)))

uint32_t *ledport;
uint32_t ledmask = 1;

void setup() {
  // Set up the built-in LED pin as an output:
  pinMode(led_pin, OUTPUT);

  while (!Serial && millis() < 4000) ;
  Serial.begin(115200);

  Serial.println((uint32_t)g_Pin2PortMapArray[led_pin].GPIOx_Port, HEX);
  Serial.println((uint32_t) & (g_Pin2PortMapArray[led_pin].GPIOx_Port->ODR), HEX);
  uint16_t GPIO_Pin = g_Pin2PortMapArray[led_pin].Pin_abstraction;
  Serial.println(GPIO_Pin, HEX);
  uint8_t pin_number = 0;
  while (GPIO_Pin != 1) {
    GPIO_Pin >>= 1;
    pin_number++;
  }
  Serial.println(pin_number, DEC);

  ledport = GPIO_BITBAND_PTR(g_Pin2PortMapArray[led_pin].GPIOx_Port->ODR, pin_number);
  Serial.println((uint32_t)ledport, HEX);
}

void loop() {
  *ledport |= ledmask;      // digitalWrite(HIGH);
  delay(250);
  *ledport &= ~ledmask;        // digitalWrite(LOW);
  delay(250);
}
```
How best to set this up?  With the Teensy boards, There is a table with mapping pin number to port register (which if I remember correctly is both input and output, where here ODR/IDR... 

But looks like it is all doable!

Side note: with functions like digitalWrite, wonder if there should be some verification that the pin number is in range.  Example what happens if you call digitalWrite(100, HIGH);
The code will try to index into the g_Pin2PortMapArray[ulPin] array and...

KurtE, 2018-07-28:
@OpusK 
For the fun of it I played around with trying to add these using bitband.  So I edited my fork/new branch, and for the heck of it added in the digitalWriteFast stuff we added to OpenCR... 

Then defined a table bitband addresses for all of the GPIO pins, and defined the macros.  Might need additional work, but put it up in PR, so you could take a look.
https://github.com/ROBOTIS-GIT/OpenCM9.04/pull/29

The Pull request has example app, that blinks the LED a few different ways.  digitalWriteFast, 
the output port method, and a portSetRegister/portClearRegister call... 

I still need to test the input register and I have not tried implementing one to change direction, as not sure there is an easy way to do this.

Let me know what you think.

OpusK, 2018-07-31:
@KurtE,

That's awesome!
I just finished your PR. I will commit some of the necessary content.

Thanks again your contribution!

OpusK, 2018-08-09:
Hi, @brycebeddard 

We have just launch a new release 1.2.0. It should solve your issue. 
Please give me your feedback.

OpusK, 2018-08-20:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-08-20 by OpusK

brycebeddard, 2018-08-27:
Apologies, I was away from work for a few weeks and was unable to check the new release.

After attempting to compile the "graphicstest" example from Adafruit's ST7735 library, I encounter a singular error.

`In file included from C:\Users\intern-04\Documents\ArduinoData\packages\OpenCM904\hardware\OpenCM904\1.2.0\cores\arduino/Arduino.h:23:0,

                 from sketch\graphicstest.ino.cpp:1:

c:\users\intern-04\documents\arduinodata\packages\opencm904\tools\opencm_gcc\5.4.0-2016q2\lib\gcc\arm-none-eabi\5.4.1\include\stdint.h:9:26: fatal error: stdint.h: No such file or directory

compilation terminated.`

Will be investigating this further myself. Thanks for the work you've put into 1.2.0!