ROBOTIS-GIT/OpenCM9.04 #26: Should digitalWrite - only work with HIGH and LOW?

KurtE, 2018-07-22:
I thought I would try out one of the forum posts about a program not working on Arduino that worked on other SDK... 
http://en.robotis.com/service/forum_view.php?bbs_no=2406903&page=1&save_sca=&sca=&save_stx=&stx=&sfl=

So compiled program and looked at output with Logic Analyzer.  The data did not look valid... 
The data was output using the line:
```
 digitalWrite(Max7219_pinDIN,DATA&0x80)
```
Which was not working.  However the line:  
```
digitalWrite(Max7219_pinDIN,(DATA&0x80)? HIGH : LOW);
```
Took a quick look at the digitalWrite code: 
```
extern void digitalWrite( uint32_t ulPin, uint32_t ulVal )
{

  switch(ulVal) {

    case HIGH:
      HAL_GPIO_WritePin(g_Pin2PortMapArray[ulPin].GPIOx_Port,g_Pin2PortMapArray[ulPin].Pin_abstraction,GPIO_PIN_SET);
			break;

    case LOW:
      HAL_GPIO_WritePin(g_Pin2PortMapArray[ulPin].GPIOx_Port,g_Pin2PortMapArray[ulPin].Pin_abstraction,GPIO_PIN_RESET);
			break;

    default:
			break;
  }
}
```
It looks like it only works with the two values HIGH and LOW.  Most systems I have worked with assumed HIGH for any non 0 value.

OpusK, 2018-07-23:
Hi, @KurtE ,
Thanks for your report.
You're right! It should be fixed. We will fix this in the next release.

OpusK, 2018-07-24:
I have updated the code in the develop branch.

closed 2018-07-26 by OpusK