ROBOTIS-GIT/OpenCM9.04 #27: Arduino builds on linux arm boards (RPI, ODroid, ...)

KurtE, 2018-07-24:
Sometimes I find it convenient to be able to do simple Arduino builds for boards like the OpenCM board while it is plugged into an ARM based SBC, like the RPI3 and the Odroid XU4 or... 

You can install Arduino to run on these boards, and you can follow the instructions to install the OpenCM board.  

Yes I know that these boards are a lot slower to build on, but sometimes it beats having to dig in to where the OpenCM board is, unplug it, move it to your PC (Windows, Linux, MAC...), reprogram it, than move it back to the SBC. 

Note: Most of the time I do the builds using the GUI.  I start up a VNCServer on the RPI3 in this case, then open a viewer on my PC... 

Today I tried this, and the compile worked, but the download fails. 
```"/home/kurt/.arduino15/packages/OpenCM904/tools/opencm_gcc/5.4.0-2016q2/bin/arm-none-eabi-objcopy" -O binary  "/tmp/arduino_build_790505/OpenCM_Servo_Controller.ino.elf" "/tmp/arduino_build_790505/OpenCM_Servo_Controller.ino.bin"
Using library DynamixelSDK at version 0.0.1 in folder: /home/kurt/.arduino15/packages/OpenCM904/hardware/OpenCM904/1.1.0/libraries/DynamixelSDK 
Sketch uses 77004 bytes (65%) of program storage space. Maximum is 116736 bytes.
Global variables use 11204 bytes of dynamic memory.
/home/kurt/.arduino15/packages/OpenCM904/tools/opencm_tools/0.0.2/linux/opencm9.04_ld /dev/ttyACM0 57600 /tmp/arduino_build_790505/OpenCM_Servo_Controller.ino.bin 1 opencm 
An error occurred while uploading the sketch
/home/kurt/.arduino15/packages/OpenCM904/tools/opencm_tools/0.0.2/linux/opencm9.04_ld: 1: /home/kurt/.arduino15/packages/OpenCM904/tools/opencm_tools/0.0.2/linux/opencm9.04_ld: Syntax error: end of file unexpected
```
It is trying to run the standard intel 32 bit linux image... 

Assuming you can get this to work, the next thing that would be great is if it also worked with the command line invocation of Arduino (GUI not required), than can setup to build over a putty command line.

OpusK, 2018-07-24:
Hi, @KurtE,

I think it is a good opinion.
Current loader does not support ARM architecture. I will discuss this with the team.

KurtE, 2018-07-24:
Thanks, 

Side note: I believe at some point (hopefully in near future) Arduino will also release an ARM64 version. 
It was mentioned up on the PJRC(Teensy) forum by Paul: https://forum.pjrc.com/threads/52525-Teensyduino-1-42-Beta-6?p=180442&viewfull=1#post180442

I verified one version of the Paul's ARM64 stuff on an ODroid C2 running Ubuntu Arm 64.

OpusK, 2018-07-27:
@KurtE  - First, we will build and commit to the ARM architecture.
It is very old code related to boot loader source release ... It seems to be released after some confirmation.

robotpilot, 2018-08-08:
@OpusK, Is this issue complete?

OpusK, 2018-08-09:
@robotpilot, 
Currently in progress. There is a slight problem and we are checking.

KurtE, 2018-11-07:
@OpusK  and @chcbaram 

Was wondering if you have a working version of the downloader that runs on RPI or other Linux ARM?

OpusK, 2018-11-08:
@KurtE,   

I do not have it yet. I have not been able to continue debugging this recently. I'm sorry to be late.

KurtE, 2018-11-08:
Thanks,
I was just wondering as another user up on Trossen is wanting to do wireless updates and I mentioned 
This and maybe the possibility of using something like a RPI0w to download a hex file to and then run some command line to program the board...

OpusK, 2018-11-09:
@KurtE ,

I am sorry to be unable to release Loader Sources due to company policy.
There is still a problem with serial communication. Whether this is a boot loader problem in OpenCM 9.04 or a loader code problem is a further need for debugging.

However, I attached the loader file that I tested in RaspberryPi3 B+ (with Raspbian OS).
It does not work reliably, but I think it will be better than nothing.

[opencm9.04_ld_arm32.zip](https://github.com/ROBOTIS-GIT/OpenCM9.04/files/2564722/opencm9.04_ld_arm32.zip)

KurtE, 2018-11-11:
Thanks @OpusK,

I have not had a chance to try it out, but the other person on Trossen has and says it works at least some of the time as you can see up at:
http://en.robotis.com/service/forum_view.php?bbs_no=2407088&page=1&save_sca=&sca=&save_stx=&stx=&sfl=

OpusK, 2018-11-16:
@KurtE,

Thank you for your feedback.
Since the loader is not stable yet, I will keep this issue open.

KurtE, 2018-11-16:
hi @OpusK, 

I will try to setup an RPI with it and try it out myself.   

It might be he was trying to upload when he had a program that was currently misbehaving, which requires the hold button down while processor is resetting... 

Still would be good if you maybe went ahead and setup that version for ARM, to be installed.  That is the user would be better off today than they were.   

Also I don't know if it is possible, but can you make a version that does verbose output.  That is, is there a version, (maybe it already is).  that maybe outputs more information, which maybe gives you some more clues. 

Things like: Did it find the device,  Could it reset the device,  Did it have write access... 

Again I will try to play with it over the next few days

KurtE, 2018-11-16:
@OpusK 

I now have Arduino 1.8.7 installed on RPI3 with your ROS2 installation...  updated the udev rules...

I then edited the preferences and had it install the OpenCM board... I then downloaded your updated loader file into the place that Arduino runs it from

It took a couple of attempts, including force the board into boot loader mode.   Finally got it to download, but it is not blinking the LED...   I tried building blink program on windows machine and it blinks. I tried to copy the resulting file down to RPI and ran it using the loader you have and took a couple of attempts and it succeeded, but no blink... 

Then I also git cloned my fork of OpenCM9.04 and did the stuff to setup to be able to build using the develop branch.

I tried building blink using the develop branch and get compiler errors: 
```
/home/pi/.arduino15/packages/OpenCM904/tools/opencm_gcc/5.4.0-2016q2/bin/arm-none-eabi-g++ -c -g -O2 -std=gnu++11 -Wno-write-strings -MMD -ffunction-sections -fdata-sections -fno-rtti -fno-exceptions -DARM_MATH_CM3 -DUSE_HAL_DRIVER -DBOARD_OpenCM904 -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10807 -DARDUINO_OpenCM904 -DARDUINO_ARCH_OPENCM904 -mthumb -DSTM32F103xB -D__OPENCM904__ -I/home/pi/Arduino/hardware/opencm904/OpenCM904/variants/OpenCM904/bsp/opencm -I/home/pi/Arduino/hardware/opencm904/OpenCM904/variants/OpenCM904/bsp/opencm/include -I/home/pi/Arduino/hardware/opencm904/OpenCM904/variants/OpenCM904/hw -I/home/pi/Arduino/hardware/opencm904/OpenCM904/variants/OpenCM904/hw/driver -I/home/pi/Arduino/hardware/opencm904/OpenCM904/variants/OpenCM904/hw/usb_cdc -I/home/pi/Arduino/hardware/opencm904/OpenCM904/variants/OpenCM904/lib/STM32F1xx_HAL_Driver/Inc/ -I/home/pi/Arduino/hardware/opencm904/OpenCM904/variants/OpenCM904/ -I/home/pi/Arduino/hardware/opencm904/OpenCM904/cores/arduino -I/home/pi/Arduino/hardware/opencm904/OpenCM904/variants/OpenCM904 /home/pi/Arduino/hardware/opencm904/OpenCM904/cores/arduino/WMath.cpp -o /tmp/arduino_build_324753/core/WMath.cpp.o
/home/pi/Arduino/hardware/opencm904/OpenCM904/cores/arduino/WString.cpp: In constructor 'String::String(unsigned char, unsigned char)':
/home/pi/Arduino/hardware/opencm904/OpenCM904/cores/arduino/WString.cpp:74:23: error: 'utoa' was not declared in this scope
  utoa(value, buf, base);
                       ^
/home/pi/Arduino/hardware/opencm904/OpenCM904/cores/arduino/WString.cpp: In constructor 'String::String(unsigned int, unsigned char)':
/home/pi/Arduino/hardware/opencm904/OpenCM904/cores/arduino/WString.cpp:90:23: error: 'utoa' was not declared in this scope
  utoa(value, buf, base);
                       ^
/home/pi/Arduino/hardware/opencm904/OpenCM904/cores/arduino/WString.cpp: In member function 'unsigned char String::concat(unsigned int)':
/home/pi/Arduino/hardware/opencm904/OpenCM904/cores/arduino/WString.cpp:307:19: error: 'utoa' was not declared in this scope
  utoa(num, buf, 10);
                   ^
exit status 1
Error compiling for board OpenCM9.04 Board.
```

KurtE, 2018-11-17:
I closed #56 above as I had not fully synced with your latest changes including the lib_f103.a file, so showed conflict.  Redid it again with pull #57

OpusK, 2018-11-19:
@KurtE ,

Thanks for your contribution!
There seems to be a problem depending on the version you are using. (I did not have any problems with the RPi3 I tested)
Anyway, I'm glad that you can do it in Teensy's way :)

I will merge your PR(#57).

closed 2019-03-14 by OpusK