ROBOTIS-GIT/OpenCM9.04 #31: SampleCode

Vali18, 2018-08-15:
Hi @KurtE @OpusK @ROBOTIS-Will  
I am trying to write a basic code for AX-12a : wheel mode writing speed and reading position. I would like to create a base and then add other functionalities(like reading load etc).
The main thing is that I would like to use the standard arduino structure:
Setup() - code to run once 
Loop() - main code to run repeatedly

This is not working...any idea?

#include <DynamixelSDK.h>

#define AX_CW_ANGLE_LIMIT_H         7
#define AX_CCW_ANGLE_LIMIT_L        8
#define AX_CCW_ANGLE_LIMIT_H        9
/** RAM AREA **/
#define AX_TORQUE_ENABLE            24
#define AX_GOAL_POSITION_L          30
#define AX_GOAL_POSITION_H          31
#define AX_GOAL_SPEED_L             32
#define AX_GOAL_SPEED_H             33
#define AX_TORQUE_LIMIT_L           34
#define AX_TORQUE_LIMIT_H           35
#define AX_PRESENT_POSITION_L       36
#define AX_PRESENT_POSITION_H       37
#define AX_PRESENT_SPEED_L          38
#define AX_PRESENT_SPEED_H          39
#define AX_PRESENT_LOAD_L           40
#define AX_PRESENT_LOAD_H           41

// Protocol version
#define PROTOCOL_VERSION                1.0 

// Default setting
#define DXL_ID                          1                // Dynamixel ID: 1
#define BAUDRATE                        1000000
#define DEVICENAME                      "3"                 // Check which port is being used on your controller

#define TORQUE_ENABLE                   1                   // Value for enabling the torque
#define TORQUE_DISABLE                  0                   // Value for disabling the torque

#define ESC_ASCII_VALUE                 0x1b

dynamixel::PortHandler *portHandler;
dynamixel::PacketHandler *packetHandler; 

// Communication result
int dxl_comm_result = COMM_TX_FAIL;             
uint8_t dxl_error = 0;                     
int16_t ax_present_position = 0;
uint16_t ax_moving_speed = 1024 + 800;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  while (!Serial);
  Serial.println("Start..");

  dynamixel::PortHandler *portHandler = dynamixel::PortHandler::getPortHandler(DEVICENAME);
// Initialize PacketHandler instance
// Set the protocol version
// Get methods and members of Protocol1PacketHandler or Protocol2PacketHandler
dynamixel::PacketHandler *packetHandler = dynamixel::PacketHandler::getPacketHandler(PROTOCOL_VERSION);

  // Open port
  if (portHandler->openPort())
  {
    Serial.print("Succeeded to open the port!\n");
  }
  else
  {
    Serial.print("Failed to open the port!\n");
    Serial.print("Press any key to terminate...\n");
    return;
  }

  // Set port baudrate
  if (portHandler->setBaudRate(BAUDRATE))
  {
    Serial.print("Succeeded to change the baudrate!\n");
  }
  else
  {
    Serial.print("Failed to change the baudrate!\n");
    Serial.print("Press any key to terminate...\n");
    return;
  }
/*
  // Enable Dynamixel Torque
  dxl_comm_result = packetHandler->write1ByteTxRx(portHandler, DXL_ID, AX_TORQUE_ENABLE, TORQUE_ENABLE, &dxl_error);
  if (dxl_comm_result != COMM_SUCCESS)
  {
    packetHandler->getTxRxResult(dxl_comm_result);
  }
  else if (dxl_error != 0)
  {
    packetHandler->printRxPacketError(dxl_error);
  }
  else
  {
    Serial.print("Dynamixel has been successfully connected!\n");
  }
 */ 
  // Change Dynamixel CW Angle Limit
  dxl_comm_result = packetHandler->write2ByteTxRx(portHandler, DXL_ID, AX_CW_ANGLE_LIMIT_L,0, &dxl_error);
  if (dxl_comm_result != COMM_SUCCESS)
  {
    Serial.printf("%sn", packetHandler->getTxRxResult(dxl_comm_result));
  }
  else if (dxl_error != 0)
  {
    Serial.printf("%sn", packetHandler->getRxPacketError(dxl_error));
  }
  else
  {
    Serial.printf("Dynamixel CW angle set to 0 successfully!\n");
  }
  
  // Change Dynamixel CCW Angle Limit
  dxl_comm_result = packetHandler->write2ByteTxRx(portHandler, DXL_ID, AX_CCW_ANGLE_LIMIT_L, 0, &dxl_error);
  if (dxl_comm_result != COMM_SUCCESS)
  {
    Serial.printf("%sn", packetHandler->getTxRxResult(dxl_comm_result));
  }
  else if (dxl_error != 0)
  {
    Serial.printf("%sn", packetHandler->getRxPacketError(dxl_error));
  }
  else
  {
    Serial.printf("Dynamixel CCW angle set to 0 successfully!\n");
  }

}

void loop() {   
  // Write goal speed
  dxl_comm_result = packetHandler->write2ByteTxRx(portHandler, DXL_ID, AX_GOAL_SPEED_H, ax_moving_speed , &dxl_error);
  if (dxl_comm_result != COMM_SUCCESS)
  {
    Serial.printf("%sn", packetHandler->getTxRxResult(dxl_comm_result));
  }
  else if (dxl_error != 0)
  {
    Serial.printf("%sn", packetHandler->getRxPacketError(dxl_error));
  }

  // Read present Position
  dxl_comm_result = packetHandler->read2ByteTxRx(portHandler, DXL_ID, AX_PRESENT_POSITION_H, ax_present_position , &dxl_error);
  if (dxl_comm_result != COMM_SUCCESS)
  {
    Serial.printf("%sn", packetHandler->getTxRxResult(dxl_comm_result));
  }
  else if (dxl_error != 0)
  {
    Serial.printf("%sn", packetHandler->getRxPacketError(dxl_error));
  }

  Serial.printf("[ID:%03d] Pres:%dnn", DXL_ID, ax_present_position); 
}

KurtE, 2018-08-16:
@Vali18, @ROBOTIS-Will and @OpusK: 

Thought I would jump in here as I was mentioned and I have a working version of your code. 
There are/were lots of issues.  Some of which were mentioned on the forum.  

Things like:
As @ROBOTIS-Will  mentioned online you used the wrong registers, I missed that one when I caught the earlier errors like reading/writing 4 bytes instead of 2

Also your printf statements have issues, like: 
```
Serial.printf("%sn", packetHandler->getRxPacketError(dxl_error));
```
Should be:
```
Serial.printf("%s\n", packetHandler->getRxPacketError(dxl_error));
```
The \n is used to say print out a newline (i.e get to the next line on output)

You have the Torque Enable code commented out.   Which I put back.  Actually I moved it later in the code as with AX servos it is not important but if you move move to some of the newer servos you can not update areas in EEPROM when torque is enabled on the servo.

Probably the biggest issue I ran into was you defined global variables for the portHandler and packetHandler 
```
dynamixel::PortHandler *portHandler;
dynamixel::PacketHandler *packetHandler;
```

And then in setup you had local versions of these variables: 
and then locally in setup you defined (and set them)
```
ynamixel::PortHandler *portHandler = dynamixel::PortHandler::getPortHandler(DEVICENAME);
// Initialize PacketHandler instance
// Set the protocol version
// Get methods and members of Protocol1PacketHandler or Protocol2PacketHandler
dynamixel::PacketHandler *packetHandler = dynamixel::PacketHandler::getPacketHandler(PROTOCOL_VERSION);
```
So the global variables were never set, so when you reference them in loop(), you are indirecting off a NULL pointer and fault. 

Here is a version of your program that is working for me.  Note: I have a different SERVO ID...
```
#include <DynamixelSDK.h>

/** EEPROM AREA **/
#define AX_MODEL_NUMBER_L           0
#define AX_MODEL_NUMBER_H           1
#define AX_VERSION                  2
#define AX_ID                       3
#define AX_BAUD_RATE                4
#define AX_RETURN_DELAY_TIME        5
#define AX_CW_ANGLE_LIMIT_L         6
#define AX_CW_ANGLE_LIMIT_H         7
#define AX_CCW_ANGLE_LIMIT_L        8
#define AX_CCW_ANGLE_LIMIT_H        9
#define AX_SYSTEM_DATA2             10
#define AX_LIMIT_TEMPERATURE        11
#define AX_DOWN_LIMIT_VOLTAGE       12
#define AX_UP_LIMIT_VOLTAGE         13
#define AX_MAX_TORQUE_L             14
#define AX_MAX_TORQUE_H             15
#define AX_RETURN_LEVEL             16
#define AX_ALARM_LED                17
#define AX_ALARM_SHUTDOWN           18

/** RAM AREA **/
#define AX_TORQUE_ENABLE            24
#define AX_LED                      25
#define AX_CW_COMPLIANCE_MARGIN     26
#define AX_CCW_COMPLIANCE_MARGIN    27
#define AX_CW_COMPLIANCE_SLOPE      28
#define AX_CCW_COMPLIANCE_SLOPE     29
#define AX_GOAL_POSITION_L          30
[... 130 lines cut ...]
  {
    Serial.printf("Goal %s\n", packetHandler->getTxRxResult(dxl_comm_result));
  }
  else if (dxl_error != 0)
  {
    Serial.printf("Goal %s\n", packetHandler->getRxPacketError(dxl_error));
  }

  // Read present Position
  dxl_comm_result = packetHandler->read2ByteTxRx(portHandler, DXL_ID, AX_PRESENT_POSITION_L, &ax_present_position , &dxl_error);
  if (dxl_comm_result != COMM_SUCCESS)
  {
    Serial.printf("Present %s\n", packetHandler->getTxRxResult(dxl_comm_result));
  }
  else if (dxl_error != 0)
  {
    Serial.printf("Present %s\n", packetHandler->getRxPacketError(dxl_error));
  }

  Serial.printf("[ID:%03d] Pres:%d\n", DXL_ID, ax_present_position);
  delay(250);
}
```
Hope that helps
Kurt

Vali18, 2018-08-16:
Thank you so much @KurtE  this is awsome! Couldn't have done it without your support. Thank you!!!

I have tried the code and wheel mode works, still having  issues reading...not sure why. But so far very happy that things are finally strating to be clear and I am planning to spend sometime on it, based on all the very useful info you guys gave me... I will let u know how goes !!

Thanks a lot and please feel free to send any examples or documentation across if you find any in the future :)

KurtE, 2018-08-16:
Not sure what issues you are having reading.  I was reading in data fine. 

Note: The end of my loop had a delay of 250 or a quarter of a second.  I put this in when I was debugging and it hung system out to dry.  This is when I found that NULL pointer issue I mentioned.  

You might remove it. 

Also remember with the AX servos the values for position go 0-1091 and this only covers 300 degrees of position.  That is what is nice with the newer servos like the XL430-w250 it has almost 4 times the resolution.  The values go from 0-4095 and it covers 360 degrees.   And about the same price as the older AX-12 servos

KurtE, 2018-08-18:
@Vali18, 
I forgot to mention, that again you should probably look at the E-Manuel for some of this:
Example if you look at the [ Present Position](http://emanual.robotis.com/docs/en/dxl/ax/ax-12a/#present-position-36) you will notice in the text a comment, 

> CAUTION : If it is set to Wheel Mode, the value cannot be used to measure the moving distance and the rotation frequency.

Which again if you move up to the XL servo might work better for you...

Vali18, 2018-08-19:
Hi @KurtE yes I read this in the emanual...but when I used the workbench library I was able to read position values anyway, so I don't understand why not using SDK. I will  try more things -  like you suggested, reading only when a new value is read (?). I will definitely consider the XL for next stage of the project, but now we already have many AX12a...

Ultimately I need to count the number of rotations.  My costum robot is moving around and we need to know where is in the physical space, so the number of rotation correspond to a position in the physical system. I need to add a plus one for every full rotation. 
Is there any other way you would do it, if not reading position?

KurtE, 2018-08-19:
@Vali18 and @OpusK,

You can read the values, but again the reliability may not be very high on the values especially if you are doing other things...

Here is yet another hacked up version:
```
#include <DynamixelSDK.h>

/** EEPROM AREA **/
#define AX_MODEL_NUMBER_L           0
#define AX_MODEL_NUMBER_H           1
#define AX_VERSION                  2
#define AX_ID                       3
#define AX_BAUD_RATE                4
#define AX_RETURN_DELAY_TIME        5
#define AX_CW_ANGLE_LIMIT_L         6
#define AX_CW_ANGLE_LIMIT_H         7
#define AX_CCW_ANGLE_LIMIT_L        8
#define AX_CCW_ANGLE_LIMIT_H        9
#define AX_SYSTEM_DATA2             10
#define AX_LIMIT_TEMPERATURE        11
#define AX_DOWN_LIMIT_VOLTAGE       12
#define AX_UP_LIMIT_VOLTAGE         13
#define AX_MAX_TORQUE_L             14
#define AX_MAX_TORQUE_H             15
#define AX_RETURN_LEVEL             16
#define AX_ALARM_LED                17
#define AX_ALARM_SHUTDOWN           18

/** RAM AREA **/
#define AX_TORQUE_ENABLE            24
#define AX_LED                      25
#define AX_CW_COMPLIANCE_MARGIN     26
#define AX_CCW_COMPLIANCE_MARGIN    27
#define AX_CW_COMPLIANCE_SLOPE      28
#define AX_CCW_COMPLIANCE_SLOPE     29
#define AX_GOAL_POSITION_L          30
#define AX_GOAL_POSITION_H          31
#define AX_GOAL_SPEED_L             32
#define AX_GOAL_SPEED_H             33
#define AX_TORQUE_LIMIT_L           34
#define AX_TORQUE_LIMIT_H           35
#define AX_PRESENT_POSITION_L       36
#define AX_PRESENT_POSITION_H       37
#define AX_PRESENT_SPEED_L          38
#define AX_PRESENT_SPEED_H          39
#define AX_PRESENT_LOAD_L           40
#define AX_PRESENT_LOAD_H           41
#define AX_PRESENT_VOLTAGE          42
#define AX_PRESENT_TEMPERATURE      43
#define AX_REGISTERED_INSTRUCTION   44
#define AX_PAUSE_TIME               45
#define AX_MOVING                   46
#define AX_LOCK                     47
#define AX_PUNCH_L                  48
#define AX_PUNCH_H                  49

// Protocol version
#define PROTOCOL_VERSION                1.0

// Default setting
#define DXL_ID                          21                // Dynamixel ID: 1
#define BAUDRATE                        1000000
#define DEVICENAME                      "3"                 // Check which port is being used on your controller

#define TORQUE_ENABLE                   1                   // Value for enabling the torque
#define TORQUE_DISABLE                  0                   // Value for disabling the torque

#define ESC_ASCII_VALUE                 0x1b

[... 124 lines cut ...]
}

void PrintErrorStatus(const char *psz, int dxl_comm_result, uint8_t dxl_error)
{
  if (dxl_comm_result != COMM_SUCCESS)
  {
    Serial.print(psz);
    Serial.print(" ");
    Serial.println(packetHandler->getTxRxResult(dxl_comm_result));
  }
  else if (dxl_error != 0)
  {
    Serial.print(psz);
    Serial.print(" ");
    Serial.println(packetHandler->getRxPacketError(dxl_error));
  }
}
```
What is interesting in this version, is a quick and dirty check for Serial input at the start of Loop, where you can type in a decimal number for new speed, and then and only then does it update the ax_moving_speed.  This is nice to allow you to see what different speeds do, both for actual speed stuff, but also to see how it changes direction.

It also only does outputs values if the values change. 

It should not be hard to check for things like new value on one side of 0 and the old value on the other side to say you went a full rotation.  Many of different ways to test this.  

Again the more things (that can eat up a lot of time, like Serial.prints, can make accurate reading more difficult.

KurtE, 2018-08-20:
@Vali18 and @OpusK 

I played around some more, and tried a first pass guess on calculating revolutions.   Which does not fully work.  Not sure if anyone wants to see the whole thing here...

```
#include <DynamixelSDK.h>

/** EEPROM AREA **/
#define AX_MODEL_NUMBER_L           0
#define AX_MODEL_NUMBER_H           1
#define AX_VERSION                  2
#define AX_ID                       3
#define AX_BAUD_RATE                4
#define AX_RETURN_DELAY_TIME        5
#define AX_CW_ANGLE_LIMIT_L         6
#define AX_CW_ANGLE_LIMIT_H         7
#define AX_CCW_ANGLE_LIMIT_L        8
#define AX_CCW_ANGLE_LIMIT_H        9
#define AX_SYSTEM_DATA2             10
#define AX_LIMIT_TEMPERATURE        11
#define AX_DOWN_LIMIT_VOLTAGE       12
#define AX_UP_LIMIT_VOLTAGE         13
#define AX_MAX_TORQUE_L             14
#define AX_MAX_TORQUE_H             15
#define AX_RETURN_LEVEL             16
#define AX_ALARM_LED                17
#define AX_ALARM_SHUTDOWN           18

/** RAM AREA **/
#define AX_TORQUE_ENABLE            24
#define AX_LED                      25
#define AX_CW_COMPLIANCE_MARGIN     26
#define AX_CCW_COMPLIANCE_MARGIN    27
#define AX_CW_COMPLIANCE_SLOPE      28
#define AX_CCW_COMPLIANCE_SLOPE     29
#define AX_GOAL_POSITION_L          30
#define AX_GOAL_POSITION_H          31
#define AX_GOAL_SPEED_L             32
#define AX_GOAL_SPEED_H             33
#define AX_TORQUE_LIMIT_L           34
#define AX_TORQUE_LIMIT_H           35
#define AX_PRESENT_POSITION_L       36
#define AX_PRESENT_POSITION_H       37
#define AX_PRESENT_SPEED_L          38
#define AX_PRESENT_SPEED_H          39
#define AX_PRESENT_LOAD_L           40
#define AX_PRESENT_LOAD_H           41
#define AX_PRESENT_VOLTAGE          42
#define AX_PRESENT_TEMPERATURE      43
#define AX_REGISTERED_INSTRUCTION   44
#define AX_PAUSE_TIME               45
#define AX_MOVING                   46
#define AX_LOCK                     47
#define AX_PUNCH_L                  48
#define AX_PUNCH_H                  49

// Protocol version
#define PROTOCOL_VERSION                1.0

// Default setting
#define DXL_ID                          21                // Dynamixel ID: 1
#define BAUDRATE                        1000000
#define DEVICENAME                      "3"                 // Check which port is being used on your controller

#define TORQUE_ENABLE                   1                   // Value for enabling the torque
#define TORQUE_DISABLE                  0                   // Value for disabling the torque

#define ESC_ASCII_VALUE                 0x1b

dynamixel::PortHandler *portHandler;
[... 298 lines cut ...]
[ID:021] Pres:617
[ID:021] Pres:623
[ID:021] Pres:622
[ID:021] Pres:617
[ID:021] Pres:620
[ID:021] Pres:0
Revl: 6
[ID:021] Pres:1
[ID:021] Pres:2
[ID:021] Pres:3
[ID:021] Pres:4
[ID:021] Pres:5
[ID:021] Pres:6
[ID:021] Pres:7
[ID:021] Pres:9
```

Notice that it nicely counts up to 1023, than gets several garbage values before getting back to to counting 1, 2, 3, ... 

So my above code does increment/decrement revolution counter, but several times during this transition...   So you would still need to put in some other filtering, to know how a valid count...  
Could be as simple as checking for values going up to some high value near 1023, and then ignore all values until you get one in the range of under some value like 10 or 20... Maybe verifying that it had a natural counting up before the break point and valid after break point and then increment your value... 

Or if you experiment and find the garbage values are never in some range like 300-350, than you could look for counting that goes through that range to increment your revolution count... 

Again I think you can probably get something that works pretty well.  My guess is it will be more reliable at lower speeds, but...

KurtE, 2018-08-20:
@Vali18 and (@OpusK @routiful @kijongGil )

I thought I would give it one more try,  It is sort of kludge code,  But it looks for values near 512.  If it finds a value between 500-510 and another value 513-523 and no values < 500 or > 523 in between them than it assumes it completed a revolution...   It appears to be working pretty well...
```
#include <DynamixelSDK.h>

/** EEPROM AREA **/
#define AX_MODEL_NUMBER_L           0
#define AX_MODEL_NUMBER_H           1
#define AX_VERSION                  2
#define AX_ID                       3
#define AX_BAUD_RATE                4
#define AX_RETURN_DELAY_TIME        5
#define AX_CW_ANGLE_LIMIT_L         6
#define AX_CW_ANGLE_LIMIT_H         7
#define AX_CCW_ANGLE_LIMIT_L        8
#define AX_CCW_ANGLE_LIMIT_H        9
#define AX_SYSTEM_DATA2             10
#define AX_LIMIT_TEMPERATURE        11
#define AX_DOWN_LIMIT_VOLTAGE       12
#define AX_UP_LIMIT_VOLTAGE         13
#define AX_MAX_TORQUE_L             14
#define AX_MAX_TORQUE_H             15
#define AX_RETURN_LEVEL             16
#define AX_ALARM_LED                17
#define AX_ALARM_SHUTDOWN           18

/** RAM AREA **/
#define AX_TORQUE_ENABLE            24
#define AX_LED                      25
#define AX_CW_COMPLIANCE_MARGIN     26
#define AX_CCW_COMPLIANCE_MARGIN    27
#define AX_CW_COMPLIANCE_SLOPE      28
#define AX_CCW_COMPLIANCE_SLOPE     29
#define AX_GOAL_POSITION_L          30
#define AX_GOAL_POSITION_H          31
#define AX_GOAL_SPEED_L             32
#define AX_GOAL_SPEED_H             33
#define AX_TORQUE_LIMIT_L           34
#define AX_TORQUE_LIMIT_H           35
#define AX_PRESENT_POSITION_L       36
#define AX_PRESENT_POSITION_H       37
#define AX_PRESENT_SPEED_L          38
#define AX_PRESENT_SPEED_H          39
#define AX_PRESENT_LOAD_L           40
#define AX_PRESENT_LOAD_H           41
#define AX_PRESENT_VOLTAGE          42
#define AX_PRESENT_TEMPERATURE      43
#define AX_REGISTERED_INSTRUCTION   44
#define AX_PAUSE_TIME               45
#define AX_MOVING                   46
#define AX_LOCK                     47
#define AX_PUNCH_L                  48
#define AX_PUNCH_H                  49

// Protocol version
#define PROTOCOL_VERSION                1.0

// Default setting
#define DXL_ID                          21                // Dynamixel ID: 1
#define BAUDRATE                        1000000
#define DEVICENAME                      "3"                 // Check which port is being used on your controller

#define TORQUE_ENABLE                   1                   // Value for enabling the torque
#define TORQUE_DISABLE                  0                   // Value for disabling the torque

#define ESC_ASCII_VALUE                 0x1b

dynamixel::PortHandler *portHandler;
dynamixel::PacketHandler *packetHandler;
[... 194 lines cut ...]
Revl: 7 Dt:1532
Revl: 8 Dt:1532
Revl: 9 Dt:1532
New speed 800 CCW
Revl: 10 Dt:1246
Revl: 11 Dt:1155
Revl: 12 Dt:1153
Revl: 13 Dt:1157
Revl: 14 Dt:1155
New speed 0 CCW
New speed 176 CW
Revl: 13 Dt:6093
Revl: 12 Dt:5540
New speed 376 CW
Revl: 11 Dt:5219
Revl: 10 Dt:2455
Revl: 9 Dt:2455
Revl: 8 Dt:2454
New speed 476 CW
Revl: 7 Dt:2433
Revl: 6 Dt:1921
Revl: 5 Dt:1924
New speed 0 CCW
```
I think I am pretty well done with it... Hopefully it works for you

Vali18, 2018-08-21:
Dear @KurtE this is beyond impressive! Will have access to my motors again tomorrow and it will be the first thing I'll try!!! Will let you know... And again thanks so much for all your work and support!!

Vali18, 2018-08-22:
@KurtE This works perfectly!! I have no words to thank you :)!!!!!!!!!

KurtE, 2018-08-23:
@Vali18, @OpusK ,

Glad it works for you.  My guess is this issue can probably be closed... 

Not sure it would be worth converting something like the above code into a sample program that goes with the openCM IDE...

OpusK, 2018-08-26:
@KurtE 
Thank you for your great contribution!

@Vali18, We will discuss with the person in charge of adding examples.

@kijongGil, What do you think about adding example above?

KurtE, 2018-08-26:
@OpusK @kijongGil,

Hard to say if this one warrants being included with IDE.  Not sure if there is some other location that either code or simply approach might be mentioned on how to maybe support wheel mode on AX servos and get a revolution count...