ROBOTIS-GIT/OpenCM9.04 #37: Pull Requests: develop branch what files need to be updated in ...opencm904_release?

KurtE, 2018-09-12:
Again sorry in advance.  You probably have already answered this.  

I am looking to maybe issue a new Pull request for some of the Serial updates and again wondered what files need to be updated in the opencm9.04_release directory.   I know that the lib_f103.a file needs to be updated.  But I also thought all updated header files?  What about Library source files?  and the  like.

What about if I created a keywords.txt (I have), which include some of the things that are not currently showing up as special things in the arduino ide?

Currently mine looke like:
```
# OpenCM specific functions
printf	KEYWORD2
flushRx	KEYWORD2
transmitterEnable	KEYWORD2
setDxlMode	KEYWORD2
getBaudRate	KEYWORD2
getRxCnt	KEYWORD2
getTxCnt	KEYWORD2

# Arduino constants

# advanced pin states
OUTPUT_OPEN	LITERAL1
INPUT_PULLDOWN	LITERAL1
INPUT_ANALOG	LITERAL1
```
(The transmitterEnable is new in my stuff)

I tried running WinMerge to see which files are different.  First on my current code and updated all of the header files I had changed, But there are several other files I have not touched, so I looked at the files in my clone of your fork/branch and it too shows several files different.  

Things like UARTClass.h, variant.h drv_uart.h , but in addition there are other source files, like:
port_handler_arduino.cpp, variant.cpp,  that is also different, 

And many of the example sketches where mostly the changes are changing the comment about 
changing comments: 
 http://support.robotis.com/ -> http://emanual.robotis.com/

But also a few with actual changes. 

So mainly just wondering which types of these files should be updated with a Pull request.  

Thanks
Kurt

OpusK, 2018-09-15:
@KurtE,

Sorry for late reply you.
PullRequest requires the following:
 - develop firmware in [opencm9.04 folder](https://github.com/ROBOTIS-GIT/OpenCM9.04/tree/master/arduino/opencm_arduino/opencm9.04)
 - update lib_f103.a in opencm904_release folder
 - update include files in opencm904_release folder (For changes only)

> port_handler_arduino.cpp, variant.cpp, that is also different,

These files will be applied to the .a file through the build if you have modified them in the opencm9.04 folder. Therefore, if these files have been modified, you can leave them in the opencm9.04 folder and do PullRequest.

> What about if I created a keywords.txt (I have), which include some of the things that are not currently showing up as special things in the arduino ide?

I think it's a good idea :) 

And... since our code structure does not distinguish between include and src folders, it is inconvenient to copy the modifications one by one.
We plan to separate the include and src folders in the future.

OpusK, 2018-10-04:
Hi, @KurtE.

Has this issue been resolved?

KurtE, 2018-10-04:
Yes, good time to close

closed 2018-10-04 by KurtE