ROBOTIS-GIT/OpenCM9.04 #40: USB Serial: See what Baud rate the host is asking for....

KurtE, 2018-10-03:
There are times when a sketch would want to know which baud rate the usb host asks for.  Example the Dynamixel Manager and R+ Manager sets the host baud rate, which I assume the default OpemCM9.04 firmware uses to set the DXL baud rate... 

Another example the example program c_Serial_Serial2_Converter program, would be a lot more functional if it could change the Serial2 Baud rate on the fly to whatever the host says it wants... 

It is currently possible to do this:  The base code already receives the CDC_SET_LINE_CODING message and saves the desired baud rate into the variable: LineCoding.bitrate 

So for example the Serial to Serial2 converter can today do this, maybe looking something like:
```
#define BAUDRATE 57600
extern USBD_CDC_LineCodingTypeDef LineCoding;
uint32_t g_baud_rate = BAUDRATE;
void setup(){
  Serial.begin(115200);
  Serial2.begin(BAUDRATE );
  pinMode(BOARD_LED_PIN, OUTPUT);
}

void loop(){
  if(Serial.available()){
    Serial2.print((char)Serial.read());//send data coming from USB to Serial2
  }

  if(Serial2.available()){
    toggleLED();
    Serial.print((char)Serial2.read()); //send data coming from Serial2 to USB(PC)
  }
  // See if the host asked to change baud rates
  if (LineCoding.bitrate != g_baud_rate) {
    g_baud_rate =LineCoding.bitrate;
    Serial2.begin(g_baud_rate);
  }
}
```
Warning, I typed the changes in on the fly, so may not be 100% correct, but do have simple DXL forwarder with this, which is switching baud rates. 

While the above works, I might suggest making it easier to do. 
That is currently if you do:
```
    Serial.print("USB Serial Baud: ");
    Serial.println(Serial.getBaudRate(), DEC);
```
Currently getBaudRate simply returns the value usb_cdc_bitrate 
Which I found a an extern definition for this variable and some binary file... 

When I tried printing this value out, it returned the value 536891392, which is 0x20005000.

Maybe would be better if getBaudRate returned the value from LineCoding.bitrate.

OpusK, 2018-10-04:
Hi, @KurtE,

I think it is good.
Actually, [OpenCR](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/cores/arduino/USBSerial.cpp#L101) and [CommXEL](https://github.com/ROBOTIS-GIT/XelNetwork_CommXel/blob/master/commXel_app/src/ap/usb_to_dxl/usb_to_dxl.cpp#L22) are using that function.

I think it would be nice to add functionality to OpenCM 9.04.
In this regard, we may receive your PR to make your contribution. What would you like to do?

KurtE, 2018-10-04:
I will work on a PR,  probably the most simplistic version as mentioned of updating Serial.getBaudRate to return this value. 

I will then make a version of the Serial to Serial2 example program, that allows the user to make use of it. 

Probably under #ifdef 
Example if BAUDRATE is defined, then always use it, else use whatever USB tells you...

OpusK, 2018-10-05:
@KurtE, 
Thanks for your PR!
I have just merged your PR, but it would be better if you add the author later when you add examples.
(ex. Author: KurtE)

KurtE, 2018-10-05:
@OpusK ,

You are welcome, having fun...

As for the example: it was already an existing example, I just added a little to it.  But will add comments such as you suggested to future examples. 

Not sure if this one can be closed yet, or you wait until it gets into public release.

OpusK, 2018-10-11:
Since your PR is merged, this issue will be closed.
Thanks!

closed 2018-10-11 by OpusK