ROBOTIS-GIT/OpenCM9.04 #44: OpenCM9.04 & Visual Micro Debugging

mtrossen, 2018-10-08:
Hello,

We are developing our firmware for our robotic arms to run on the CM9.04 and using the dynamixelSDK. We have recently moved over to coding in Visual Studio using the Visual Micro extension.  The visual micro extension is for developing Arduino projects inside the Visual Studio C++ IDE. It has many helpful features and allows for far more advanced development.

We are trying to set up debugging to work with the OpenCM9.04 board and I'm posting for help on how to achieve this. Offering the OpenCM9.04 to work with Visual Micro would be a huge advantage for all your 3rd party developers.

Visual Micro says that we have to set up the software serial debugger option and we are not sure of the proper settings to make this work. 

I have made a post at VM forums here:
https://www.visualmicro.com/forums/YaBB.pl?num=1538797491/0#4

Here are links explaining how to set up serial for debugging:
https://www.visualmicro.com/post/2012/05/06/Configure-debug-for-other-types-of-serial-transport.aspx

https://www.visualmicro.com/post/2013/04/26/How-to-debug-Arduino-on-a-different-usb-serial-port.aspx

Attached is a screenshot of the output shown when trying to run debugging. And on the left in the box is project settings that we need to set up correctly.

[image]

OpusK, 2018-10-11:
Hi, @mtrossen 

Visual Micro's debugging method is using a serial device. For the ARM core, DAPLink using CMSIS-DAP must be supported, but unfortunately, OpenCM 9.04 does not.
Therefore, it seems that there is no way to use Visual Micro at this time.

OpenCM supports only [ST's debugging method](http://emanual.robotis.com/docs/en/parts/controller/opencm904/#jtagswd-4-pin). 
How to debug stm32 chips in VS(not Visual Micro) is like [this link](https://visualgdb.com/tutorials/arm/stm32/).

mtrossen, 2018-10-11:
Hi @OpusK 

Thank you for the reply. I will look into this solution you have suggested.

I have a few questions: 

1) We are developing front end software to communicate with our firmware on the OpenCM9.04 over USB serial on windows PC. Would your suggested solution allow for debugging at the same time as another software is communicating with the board on USB serial? Or could it be set up to use two different serial ports like in Visual Micro?

2) Is the difference because Visual Micro is for ARM chips and the OpenCM9.04 is using different chips?

Thank you for your help!

OpusK, 2018-10-11:
@mtrossen,

1. At present, there is no official solution.
However, I think using EclipseIDE(please refer to [Sloeber](http://eclipse.baeyens.it/) for Arduino) is enough. Or it seems to be possible in the way I linked above. (Debug with ST-Link, check data with USB)

2. OpenCM 9.04 also uses ARM core. The difference is that OpenCM 9.04 does not support [DAPLink](https://os.mbed.com/handbook/DAPLink).
To support this, a separate MCU must be connected, which is not possible because it means OpenCM9.04 hardware change.

mtrossen, 2018-10-11:
@OpusK 

Thank you, I will test your advised method. It is a different IDE, but we can change to a new IDE if it allows for debugging. I will report back after we test.

VisualMicro, 2018-10-20:
Visual Micro uses serial debug but sometimes needs to know the underlying class of the Serial or a.n.o debug object. In this case adding a board.txt to your project with the following setting will solve the "Debug" build issue.

`vm.debug.class_type=VM_DEBUGGER_TYPE_USARTSERIAL`

If you need further assistance please continue the thread [here](https://www.visualmicro.com/forums/YaBB.pl?num=1538797491/0#9)

Thanks for the post and all the comments

mtrossen, 2018-10-22:
Hello,

The Visual Micro Admin is correct. VM does work with debugging on the OpenCM9.04. I got it working tonight. For anyone else who finds this thread looking to also use VM with the OpenCM9.04 you can see the thread here: 
https://www.visualmicro.com/forums/YaBB.pl?num=1538797491/0#12

mtrossen, 2018-10-22:
Hello,

The Visual Micro Admin is correct. VM does work with debugging on the OpenCM9.04. I got it working tonight. For anyone else who finds this thread looking to also use VM with the OpenCM9.04 you can see the thread [here](https://www.visualmicro.com/forums/YaBB.pl?num=1538797491/0#12).

mtrossen, 2018-10-22:
I wanted to also leave a note that you can use the dual serial port feature so that you can debug on Serial2 while a separate program is communicating through the main SerialUSB. The guide for it is here:
https://www.visualmicro.com/page/User-Guide.aspx?doc=Debugging-With-Different-Ports.html

closed 2018-10-22 by mtrossen