ROBOTIS-GIT/OpenCM9.04 #47: Issues reading data from MX-28AR

GJohns1989, 2018-10-22:
I am unable to read out any information from my MX-28AR dynamixel.

Hardware config: OpenCM9.04 - RS485 Expansion Board - MX-28AR

Running Protocol 2.0

I am unable to get any information from the dynamixel to print over the serial channel. I am able to send data to the servo and I see the response (IE, the servo MOVES to the position sent). 

Below is a modified version of the "o_Read_Write" example in the Dynamixel Workbench set.
`#include <DynamixelWorkbench.h>

#if defined(__OPENCM904__)
  #define DEVICE_NAME "3" //Dynamixel on Serial3(USART3)  <-OpenCM 485EXP
#elif defined(__OPENCR__)
  #define DEVICE_NAME ""
#endif   

#define BAUDRATE  57600
#define DXL_ID    1

DynamixelWorkbench dxl_wb;

void setup() 
{
  Serial.begin(57600);
  // while(!Serial); // If this line is activated, you need to open Serial Terminal.

  dxl_wb.begin(DEVICE_NAME, BAUDRATE);
  dxl_wb.ping(DXL_ID);

  dxl_wb.jointMode(DXL_ID);
}

void loop() 
{
  static int index = 0;
  int32_t present_position = 0;
  
  //int32_t goal_position[2] = {1000, 2000};
  
  //dxl_wb.itemWrite(DXL_ID, "Goal_Position", goal_position[index]);
  
  dxl_wb.itemWrite(DXL_ID, "Present_Position", 2000);
  present_position = dxl_wb.itemRead(DXL_ID, "Present_Position");
  SerialUSB.print(present_position);`

The servo moves to the position, but once I open up a serial monitor through the Arduino IDE, all I get is zeros...

[image]

I have tried reading values from several different items in the servo as well as any and all of the example programs that use the itemRead() command, and I am always met with a bunch of zeros.

Regards,
Greg

KurtE, 2018-10-23:
Sorry, I was just browsing issues and wondering about the line

    dxl_wb.itemWrite(DXL_ID, "Present_Position", 2000);

Guessing it should be goal position?

GJohns1989, 2018-10-23:
Yes. It should be goal

routiful, 2018-10-23:
You can't write value to `Present_Position` address. Please refer attached link.
http://emanual.robotis.com/docs/en/dxl/x/xm430-w210/#present-position

If you want to control dynamixels, you have to set `Goal_Position` address

KurtE, 2018-10-23:
Again sorry for me jumping in again...  Especially since I don't have any servos that are RS485...  

But if it were me, some of the things I would check include:

From the sketch, I am assuming you are running Arduino.  Which version and are you up to date on the Board version?  If not I would update that. 

You have a proper power supply plugged into expansion board and you have the power turned on.  Again assume this as you say the servo moves.

Try running some of the other example programs.  Example if you run t_find_Dynamixel from the Examples->OpenCM904->DynamixelWorkbench does it find your servo?   If so the hardware/software is limping along as the servo had to return a packet that says it was there... 

Try running R+Manager Version 2 (as you say you are using Protocol2).  First go up and choose Robotis Mini, with the OpenCM board plugged in, go to firmware recovery.  Go through the steps to program the OpenCM board with default firmware (Note you will probably need to put the board in firmware recovery,  mode, by holding button and pressing reset...  Once the OpenCM9.04 firmware is done,  Go to update/Test and you should be able to select the OPenCM... and bring up control table.  Make sure board is configured for your servos.  In particular: 12 Baud rate BUS, is set for your servos (probably 1 for 57600) and make sure 16 Dynamixel Channel is set to 1, for the expansion board.

Then you should be able to hit the Home button on the manager (may have to close sub window first), but then select Dynamixel as the type.  and try the Update and test and see if it finds your servo?  Hopefully it does...  If not, maybe you can try the Firmware Recovery for the servo at this point.  I don't have any of these MX servos so have not tried here... I have done some others XL/XM... 

If all of the above does not work, I would try to determine where the issue was?  Do you have more than one servo?  Do they all fail?  Or only this one?  Do you have a different controller to try the servo on?  

And because I can, if it were me, I would hook up Logic Analyzer to the DXL signals and see if the packets are being properly sent and responses coming back...  And doing some debugging that way, to again figure out if the issue is at the Servo or at the OpenCM..

Again sorry for jumping in here.  Hopefully something will work.

closed 2018-11-15 by gpwhd07