ROBOTIS-GIT/OpenCM9.04 #48: Problem with installation of library (OpenCM) for Arduino

GrigoriyAY, 2018-10-23:
I added link in preferences, but when I opened boards manager, I couldn't download library for OpenCM9.04 and I got the message (attached screenshot). PC: Windows 10 and Arduino IDE 1.8.7. Could you help me with this problem?
[image]

OpusK, 2018-10-24:
Hi, @GrigoriyAY 

Would you like to try again with the instructions below?
- [E-manual](http://emanual.robotis.com/docs/en/parts/controller/opencm904/#development-environment)

OpusK, 2018-10-31:
@GrigoriyAY 

Sorry for my late reply.

> Could you suggest how to manually place the library for OpenCM in the necessary folders?
I think there was something about this in the Arduino official document, but I can not find it now...

As an alternative, please refer to the link below.
 - [HOWTO : Adding multiple OpenCR boards to the Arduino IDE](https://community.robotsource.org/t/howto-adding-multiple-opencr-boards-to-the-arduino-ide/1497)

In this case, it is necessary to set things related to tools separately, but there is no official guide related to this.

OpusK, 2018-11-08:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-11-08 by OpusK