ROBOTIS-GIT/OpenCM9.04 #50: Problem when using more than 4 motors.

edgarcamilocamacho, 2018-10-31:
Hello. I am having a problem when I try to use more than 4 motors. The following code writes (async) DXL_CNT motors (based on some of the examples).
``` c++
#include <DynamixelWorkbench.h>

#define DEVICE_NAME "3"

#define BAUDRATE  1000000
#define DXL_CNT   4

DynamixelWorkbench dxl_wb;

int dxl_id[DXL_CNT] = {1, 2, 33, 34};

int32_t pos = 0;

void setup() {
  Serial.begin(57600);
  digitalWrite(LED_BUILTIN, HIGH);
  while(!Serial);
  digitalWrite(LED_BUILTIN, LOW);
  Serial.println("Starting...");
  delay(200);
  pinMode(LED_BUILTIN, OUTPUT);
  dxl_wb.begin(DEVICE_NAME, BAUDRATE);
  
  for(int i=0; i<DXL_CNT; i++){
    Serial.print("Ping ID=");
    Serial.print(dxl_id[i]);
    Serial.print("... ");
    dxl_wb.ping(dxl_id[i]);
    Serial.print("Done... ");
    dxl_wb.jointMode(dxl_id[i]);
    Serial.println("Connected.");
  }
}

void loop(){
  Serial.print("Pos = ");
  Serial.println(pos);
  for(int i=0; i<DXL_CNT; i++){
    dxl_wb.goalPosition(dxl_id[i], pos);
  }
  pos = pos < 500 ? pos + 100 : 0;
  delay(2000);
}
```
It works fine, the serial output is:
```
Starting...
Ping ID=1... Done... Connected.
Ping ID=2... Done... Connected.
Ping ID=33... Done... Connected.
Ping ID=34... Done... Connected.
Pos = 0
Pos = 100
Pos = 200
Pos = 300
Pos = 400
Pos = 500
Pos = 0
```
But when I change `#define DXL_CNT   4` by `#define DXL_CNT   5` and `int dxl_id[DXL_CNT] = {1, 2, 33, 34};` by `int dxl_id[DXL_CNT] = {1, 2, 33, 34, 80};`, I get:
```
Starting...
Ping ID=1... Done... Connected.
Ping ID=2... Done... Connected.
Ping ID=33... Done... Connected.
Ping ID=34... Done... 
```
And the OpenCM 9.04 processor dies, the USB connection is lost and I have to reset the board to be able to close the arduino serial monitor. 

The problem happens when I try to do jointMode the motor 34, but it worked with DXL_CNT=4!!!

The problem persists with any combination of IDs.

edgarcamilocamacho, 2018-11-03:
I tested it with the old OpenCM IDE, and it worked without problems.

routiful, 2018-11-13:
Sorry for late reply.
For the reason of memory issue, I restricted the number of series. The default is 5 and I tested it OpenCR.

https://github.com/ROBOTIS-GIT/dynamixel-workbench/blob/676880b4900877e4d6b68f21b87b0153cf04fbe3/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_driver.h#L33

I am ready for new release of Dynamixel-Workbench. It might be solved that problem.

OpusK, 2018-11-26:
This issue will be closed since there were no actions for a while.
You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-11-26 by OpusK

edgarcamilocamacho, 2018-11-28:
But it means we can now use any number of motors?