ROBOTIS-GIT/OpenCM9.04 #51: Is it possible to program barebones on this board without IDE?

Le0nX, 2018-10-31:
Hello! My question is: is it possible to program this board barebones? For example, if I want to  start my OS on it?

Le0nX, 2018-11-05:
Hey?

OpusK, 2018-11-06:
Hi, @Le0nX 

Sorry for late reply.
Because this board is low-end, it can not run a normal OS. It would be possible to have an optimized embedded OS, but generally we do not use the OS on this board.

Le0nX, 2018-11-06:
Ok! And how I can flash my openCM board (without robotis IDE) to start running my optimized OS?

Le0nX, 2018-11-08:
Thanks! Can I do it via USB or jtag is needed ?

OpusK, 2018-11-08:
Both are possible.
Since OpenCM 9.04 uses ST's chip, you will find the information associated with it and use that method
(ex: USB: dfu, JTAG: ST-LINK)

OpusK, 2018-11-16:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-11-16 by OpusK

Ash-S, 2020-02-12:
Hi, how can i flash an application to openCM904 without using arduino/openCM IDE ? Is there any way, please suggest! Thanks!