ROBOTIS-GIT/OpenCM9.04 #52: I2C OpenCM9.04C board using Arduino IDE

TahaShafa, 2018-10-31:
I'm attempting to connect to an IMU (Adafruit BNO055) using an OpenCM9.04 board with I2C. I tried using the Wire.h library and the I2CDev library, but it still doesn't recognize the connection. I have the wires connected to pins 24 and 25 for SDA and SCL connections. At the beginning of my code, I tried using wire.begin(14,15) and wire.begin(24,25) to initialize the I2C but it doesn't seem to work. Any idea how I can connect with the IMU? I know the IMU works because I can get it working for an Arduino Uno, but not for the OpemCM9.04.

OpusK, 2018-11-06:
Hi, @TahaShafa 

I can not check the operation because I do not have that hardware.

Wire.h in OpenCM 9.04 uses software i2c, which can cause this problem because of the slow speed.

In the case of I2CDev, it is an API that is incompatible with the Wire example code. Unless you modify the code to match the I2CDev API, it will not be available.

OpusK, 2018-11-16:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-11-16 by OpusK

YifanAtBioRoboLab, 2019-09-17:
Hi There,
I just happen to see this closed issue. Actually I have just encountered the exact same issue and I think I should put my solution here as a clue.

The main reason openCM9.04 and BNO055 cannot work together correctly is about the clock stretching. The BNO055 will pull down the SCL to low before and after the ACK signal for about 120 microseconds each. The hardware I2C of other devices can detect such a scenario and hold communication during that time. Yet the SlowSoftI2CMaster lib is lacking such ability, continuing sending the clock signal during that time which would not show up because of the SCL pulling low. And therefore the whole signal sequence afterward would be messed up because of such clock missing.

My simple solution is in SlowSoftI2CMaster.cpp, below L98, L104, L120, L126, adding the following code:
`while(digitalRead(_scl)!=HIGH);`
This line will do the job to detect the scenario I mentioned before and confirm the clock signal's integrity. You can change this code to some other detection methods with a timeout if there do have a possibility that something can pull the SCL to low for an unacceptable amount of time.

iicfcii, 2022-08-19:
I was also trying to use OpenCM9.04 with Adafruit BNO055. [The above solution](https://github.com/ROBOTIS-GIT/OpenCM9.04/issues/52#issuecomment-532286290) does not work for me but it is really close. Instead, I only added the following line after line 103 and line 121(both lines are `setHigh(_scl);`) in `SlowSoftI2CMaster.cpp` located at `Arduino15/packages/OpenCM904/hardware/OpenCM904/1.5.1/libraries/Wire`.  
```
while(digitalRead(_scl)!=HIGH);
```
I also noticed that if I tried to pull the data from the sensor too fast(greater than about 300Hz), it will return zeros from time to time. As long as the polling rate is not too fast. It works pretty well.