ROBOTIS-GIT/OpenCM9.04 #54: Can we detect how much stack space and heap memory is used by a sketch?

KurtE, 2018-11-15:
Not sure if this is the best place to ask?  This is not totally specific to OpenCM, this equally will apply to OpenCR, XEL, Ros To Arduino... But the actual code may differ. 

Again not sure if this is better to ask here or RobotSource or Robotis Forum? 

But wondering if there are any suggestions on how to detect how much memory is being used by a sketch.  For example I am pretty sure that when I try to blindly merge in the current Dyanmixel SDK, my test apps were failing to properly run as the stack and heap corrupted each other. 

Would be great if we could somehow find out how much non-static memory a program is using, which would would help to measure reductions in usage. 

I don't know for example if the new operation as well as malloc(), maybe call _sbrk().  If so might be able to enable test for collision... (Would need to maybe change the commented out write statement...

But again this does not show us how close we are getting... 

Was wondering if it would make sense to do something like, maybe have the startup code, do something like write a standard byte or the like to all memory over "end"  to some known quantity. 

Again assuming that _sbrk is correct and we can hack it slightly, we can hopefully find out how far the heap has grown.  

We could then scan from the heap_end back to at worst case back to current stack pointer, but basically keep walking up in memory until we hit something that is NOT our prefill values and assume the stack grew at least to there... Obviously the stack could have grown beyond that if it had used those magic values, but hopefully close enough...

Does this make sense?  Has anyone already setup something like this?

OpusK, 2018-11-16:
Hi, @KurtE 

I think it would be better to deal with this issue here than elsewhere.
Among the ROBOTIS OpenHardware, OpenCM 9.04 has the fewest resources, so I think we can do a good test here.

It would be nice to be able to measure the stack and heap usage of sketches. In an IDE such as TrueStudio, the memory area is measured and displayed in the GUI. I do not have experience with this, so I think we should try to think together.

KurtE, 2018-11-16:
Hi @OpusK @routiful @kijongGil ,

For what it is worth, I have hacked up my own AX 12_test_OpenCM sketch to try it out.  Here is my current sketch:  Note: I have also checked in the changes into my github project: https://github.com/KurtE/Open_CM_CR_Arduino_Sketches
```
//====================================================================================================
// Kurts Test program to try out different ways to manipulate the AX12 servos on the PhantomX
// This is a test, only a test...
//
// This version for Robotis OpenCM9.04
//====================================================================================================
//============================================================================
// Global Include files
//=============================================================================
#include <DynamixelSDK.h>
#include <syscalls.h>
//=============================================================================
// Options...
//=============================================================================
#if defined(__OPENCM904__)
uint8_t port_handler_numbers[] = {1, 3};    // Setup to handle both ports of openCM
#endif
#if defined(__OPENCR__)
uint8_t port_handler_numbers[] = {3};    // Setup to handle both ports of openCR
#define SERVO_POWER_ENABLE_PIN BDPIN_DXL_PWR_EN
#endif

#if defined(TEENSYDUINO)
uint8_t port_handler_numbers[] = {1};     // Default to Serial1
#if defined(__MK66FX1M0__)
// Try with the T3.6 board
#define SERVOBUS Serial1
#define SERVO_RX_PIN           27
#define SERVO_TX_PIN           26
#define SERVO_DIRECTION_PIN 28
#define SERVO_POWER_ENABLE_PIN  29
#else
#define SERVO_POWER_ENABLE_PIN 2
#endif
#endif
#define COUNT_PORTHANDLERS  sizeof(port_handler_numbers)
// Uncomment the next line if building for a Quad instead of a Hexapod.
//#define QUAD_MODE
//#define TURRET
//#define DEBUG_IO_PINS

//#define VOLTAGE_ANALOG_PIN 0
//#define SOUND_PIN 1
#define SERVO1_SPECIAL  19     // We wish to reserve servo 1 so we can see servo reset

//=============================================================================
// Define differnt robots..
//=============================================================================

// Protocol version
#define PROTOCOL_VERSION                1.0                 // See which protocol version is used in the Dynamixel
#define PROTOCOL_VERSION2                2.0                 // See which protocol version is used in the Dynamixel
#define DEVICENAME                      "3"                 // Check which port is being used on your controller
#define DXL_BAUDRATE 1000000

// Constants
/* Servo IDs */
#define     RF_COXA       2
#define     RF_FEMUR      4
#define     RF_TIBIA      6

#define     RM_COXA      14
#define     RM_FEMUR     16
#define     RM_TIBIA     18

#define     RR_COXA       8
[... 1106 lines cut ...]
Cmd: h
PrintServo Voltage called
Heap ptr: 200040e0  Usage: 3208
Stack Max: 20004698, usage: 2408
Estimated unused memory: 1464
0 - All Servos off
...
```
Now if you run it with my updated Dynamixel SDK code, which I was talking about on different thread.  
I extracted the code and then created new branch with the changes: 
https://github.com/KurtE/OpenCM9.04/tree/dynamixel_sdk_update_memory_reduction

Then run similar run, where I just enter my command: H
which does a write to enable torque.
The memory output with updated code...
```

:
Cmd: h
PrintServo Voltage called
Heap ptr: 200037d4  Usage: 892
Stack Max: 20004c10, usage: 1008
Estimated unused memory: 5180
```
Let me know what you think... Still probably needs some more sophistication, but maybe at least gives some hints...

OpusK, 2018-11-16:
@KurtE ,

```c
void printMemoryUsage()
{
  uint8_t *current_heap_ptr = (uint8_t*)_sbrk(0);
  Serial.printf("Heap ptr: %x  Usage: %d\n", (uint32_t)current_heap_ptr,
                 (uint32_t)current_heap_ptr - g_start_heap_pointer);

  // stack info
  uint8_t *sp_minus = stack_ptr - 10;  // leave a little slop
  uint8_t *p = current_heap_ptr;

  // try to find out how far the stack has been used
  while ((p < sp_minus) && (*p == 0xff)) p++;
  Serial.printf("Stack Max: %x, usage: %d\n", p, g_end_stack_pointer - (uint32_t)p);
  Serial.printf("Estimated unused memory: %d\n", (uint32_t)(p - current_heap_ptr));
}
```

The function you created below seems to be very useful.
Looking at your issue, we have planned to add a function that will return the current memory usage to OpenCR and OpenCM.
It's probably similar to the way you used it.

> Heap ptr: 200037d4  Usage: 892
> Stack Max: 20004c10, usage: 1008
> Estimated unused memory: 5180

I think this result is very meaningful.
However, with respect to the SDK, A's opinion is important, so we will need to wait until he comes back and confirms.

There seems to be quite different memory management methods in environments with a lot of resources and very few resources. However, if the API is not changed, it will not affect existing users, so it will be possible to optimize the internal code (even if it is divided into define processing)

KurtE, 2018-11-16:
Hi @OpusK,

Thanks, 
With this stuff, I think it would also be good to look at how much memory we are using in global variables as well.  So will try to add another print in memory test functions. 

Example if I compile the example b_Blink_LED it says Global variables use 9700 bytes of dynamic memory (almost half).   Question is can we cut that down more?  Example Serial1, Serial2, Serial3 all have static buffers for TX, RX of 128 bytes so maybe 768 bytes there...   I know on the Teensy, the code was setup such that each SerialX object was in it's own source file, such that only if the user used something of that Serial object, would it be included by the linker.  But I know there are tricks that need to be done to make it work, as the SerialEvent code would bring it in....  ...

As a comparison, if I compile blink on a Teensy LC global variables take up 2048 out of 8k (M0 processor) or if I comple for the T3.2 it takes up 3436 out of 64K (M3 processor). 

As for my test case in previous post.  Yes, I totally understand this is more for  @kijongGil and part of my other issue, that I put up against the library. 
https://github.com/ROBOTIS-GIT/DynamixelSDK/issues/258

I did not change any of the APIs in this comparison.  What I did do was to again change the underlying code, such that if I did things like  read or write 1, 2, 4 bytes, I did not malloc the TX buffer, but instead built it on the stack (Passed in pointer to buffer to use).  I know from the sizes that for 1, 2 bytes no need for space for stuffing, for 4 bytes, I added an extra byte to buffer in case of stuffing.   

Currently I have separate code that does the stuffing here, while it is directly outputting the data to the buffer.   Now that I have a more optimal stuffing function, may simply want to call it.  However in those cases, I would like it to not realloc buffer, but know that there was enough space allocated in the first place.    Also wondering if it would be too slimy to have the addStuffing call not call realloc if the memory did not come from heap...  That is if it is in the range: &end and _sbrk(0) 

Again more for other issue.  But again would like to get rid of realloc altogether as it is playing with fire.  Example: suppose you call malloc(32) bytes and you get txBuffer pointer, and then later call realloc(38), and some other things were allocated in the space right after where txBuffer was.  realloc will allocate memory in different location in heap and copy your 32 bytes into it, and then return the new pointer to you AND release the old memory buffer.  But then you return from this call, and the caller continues to use the OLD location and tries to write that data out, and then it frees the old memory, which typically will cause the heap to be corrupted.    But again more for the other issue.

KurtE, 2018-11-17:
@OpusK,

Not sure who in this list, but thought I would do a quick test to get an idea of where all of the space in the data section is... So I again compiled the example blink.  This time I used the default Arduino one, only edited to change which pin number.  This one does not output text to Serial.

The compiler still said:
```
Global variables use 9700 bytes of dynamic memory.
```

So I found the objdump program, I then edited the file, extracted the areas above and below data section, used grep to find lines with < then edited again removed a few bad lines, changed space to , and imported into excel, where I then used function to convert hex values to decimal, and then subtract the line below from the line, to get an idea of which items are taking space...
sort of like:
```
C:\Users\kurte\AppData\Local\Arduino15\packages\OpenCR\tools\opencr_gcc\5.4.0-2016q2\arm-none-eabi\bin\objdump.exe -D Blink.ino.elf > foo
subl foo 
grep "<" foo > data_sections.csv
subl data_sections.csv
```

Then to show the data here, I exported again to CSV file, changed the ,'s into |  and ... 
Here is the data... 

|Address|Name|Address DEC|Size|
|-------|----------------|-----------|----|
|20000000|_sdata|536870912|4|
|20000004|SystemCoreClock|536870916|4|
|20000008|huart_inst|536870920|12|
|20000014|USBD_CDC_CfgFSDesc|536870932|68|
|20000058|USBD_CDC|536871000|56|
|20000090|USBD_CDC_DeviceQualifierDesc|536871056|12|
|2000009c|USBD_CDC_OtherSpeedCfgDesc|536871068|68|
|200000e0|USBD_CDC_CfgHSDesc|536871136|68|
|20000124|JUMP_BOOT_STR|536871204|4|
|20000128|h_cdc_tx_timer|536871208|4|
|2000012c|LineCoding|536871212|8|
|20000134|USBD_CDC_fops|536871220|16|
|20000144|VCP_Desc|536871236|28|
|20000160|USBD_StringSerial|536871264|28|
|2000017c|impure_data|536871292|96|
|200001dc|_impure_ptr|536871388|4|
|200001e0|lconv|536871392|56|
|20000218|_sbss|536871448|4|
|2000021c|object.8541|536871452|24|
|20000234|heap_end.5918|536871476|4|
|20000238|DataVar|536871480|2|
|2000023a|IsInit|536871482|2|
|2000023c|VirtAddVarTab|536871484|1024|
|2000063c|I2cHandle|536872508|120|
|200006b4|is_dxl_port|536872628|4|
|200006b8|is_uart_mode|536872632|4|
|200006bc|drv_uart_rx_buf_tail|536872636|12|
|200006c8|drv_uart_num|536872648|4|
|200006cc|drv_uart_rx_buf|536872652|768|
|200009cc|sw_timer_mcros|536873420|4|
|200009d0|swtimer_tbl|536873424|160|
|20000a70|excute.8929|536873584|2|
|20000a72|sw_timer_handle_index|536873586|2|
|20000a74|sw_timer_counter|536873588|4|
|20000a78|IwdgHandle|536873592|16|
|20000a88|hw_pin_config|536873608|30|
|20000aa6|ifalt.8894|536873638|1|
|20000aa7|is_opened|536873639|1|
|20000aa8|rxd_length|536873640|4|
|20000aac|rxd_BufPtrIn|536873644|4|
|20000ab0|UserTxBufPtrIn|536873648|4|
|20000ab4|usb_cdc_bitrate|536873652|4|
|20000ab8|rxd_BufPtrOut|536873656|4|
|20000abc|is_tx_full|536873660|1|
|20000abd|is_reopen|536873661|3|
|20000ac0|UserTxBufPtrOut|536873664|4|
|20000ac4|CDC_Reset_Status|536873668|4|
|20000ac8|remotewakeupon|536873672|4|
|20000acc|cfgidx.8899|536873676|4|
|20000ad0|uwTick|536873680|4|
|20000ad4|serial3_tx_buffer|536873684|128|
|20000b54|serial1_tx_buffer|536873812|128|
|20000bd4|Serial|536873940|36|
|20000bf8|Serial1|536873976|48|
|20000c28|Serial2|536874024|48|
|20000c58|Serial3|536874072|48|
|20000c88|serial2_tx_buffer|536874120|128|
|20000d08|__malloc_sbrk_start|536874248|4|
|20000d0c|__malloc_free_list|536874252|4|
|20000d10|USBD_Device|536874256|548|
|20000f34|hADC1|536874804|48|
|20000f64|exti_data|536874852|312|
|2000109c|hOC2|536875164|112|
|2000110c|hTIM4|536875276|60|
|20001148|hTIM2|536875336|60|
|20001184|hOC4|536875396|112|
|200011f4|hOC1|536875508|112|
|20001264|pwm_init|536875620|32|
|20001284|hTIM1|536875652|60|
|200012c0|hTIM3|536875712|60|
|200012fc|pwm_period|536875772|120|
|20001374|hOC3|536875892|112|
|200013e4|pwm_freq|536876004|120|
|2000145c|hspi2|536876124|88|
|200014b4|spi_dma|536876212|32|
|200014d4|hspi1|536876244|88|
|2000152c|hDrvTim|536876332|352|
|2000168c|hdma_rx|536876684|168|
|20001734|huart|536876852|192|
|200017f4|hdma_tx|536877044|168|
|2000189c|rxd_buffer|536877212|512|
|20001a9c|BuffLength|536877724|4|
|20001aa0|UserTxBufferForUSB|536877728|512|
|20001ca0|UserTxBuffer|536878240|512|
|20001ea0|UserRxBuffer|536878752|512|
|200020a0|hpcd|536879264|1056|
|200024c0|USBD_StrDesc|536880320|256|
|200025c0|pFlash|536880576|32|
|200025e0|errno|536880608|4|
|200025e4|end-0x4|536880612|-536880612|

KurtE, 2018-11-17:
Follow on to above - I start to look for some low hanging fruit, especially larger entries, like:  VirtAddVarTab (1024 bytes)
Which is used only in hw\drivers\drv_eeprom.c  
Defined as: static  uint16_t VirtAddVarTab[NB_OF_VAR] where NB_OF_VAR is 512.
Only thing that writes to this is: 
```
int drv_eeprom_init()
{
  uint16_t i;

  for( i=0; i<NB_OF_VAR; i++ )
  {
    VirtAddVarTab[i] = i;
  }
...
```
So it simply is init as 0,1,2... 0x1ff

Two solutions.  Could change it like:
```
static const uint16_t VirtAddVarTab[NB_OF_VAR] = 
{
  0x00, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0a, 0x0b, 0x0c, 0x0d, 0x0e, 0x0f,
  0x10, 0x11, 0x12, 0x13, 0x14, 0x15, 0x16, 0x17, 0x18, 0x19, 0x1a, 0x1b, 0x1c, 0x1d, 0x1e, 0x1f,
  0x20, 0x21, 0x22, 0x23, 0x24, 0x25, 0x26, 0x27, 0x28, 0x29, 0x2a, 0x2b, 0x2c, 0x2d, 0x2e, 0x2f,
...  
  0x1e0, 0x1e1, 0x1e2, 0x1e3, 0x1e4, 0x1e5, 0x1e6, 0x1e7, 0x1e8, 0x1e9, 0x1ea, 0x1eb, 0x1ec, 0x1ed, 0x1ee, 0x1ef,
  0x1f0, 0x1f1, 0x1f2, 0x1f3, 0x1f4, 0x1f5, 0x1f6, 0x1f7, 0x1f8, 0x1f9, 0x1fa, 0x1fb, 0x1fc, 0x1fd, 0x1fe, 0x1ff
};
```
Which I tried (#ifdef the for loop I showed above).  Compiles fine and sure enough data usage drops by 1024 bytes, but code size increases by the static data. 

Option 2: 
Could define something like: 
```
#define VIRTADDVARTAB(index)   (index)
```
Could make this conditional.  where as an option you could still define the variable and
then define it as: #define VIRTADDVARTAB(index)   VirtAddVarTab[index]

Then change the 9 places in this file, like: 
```
          if (( *(__IO uint16_t*)(PAGE0_BASE_ADDRESS + 6)) == VirtAddVarTab[varidx])
```
to
```
          if (( *(__IO uint16_t*)(PAGE0_BASE_ADDRESS + 6)) == VIRTADDVARTAB(varidx))
```
This would remove the data and probably not grow the code... Maybe even shrink it...

Looks like USB is using at least 3.5K or memory.

Each Serial1/2/3 uses at least 256 byte RX buffer 128 byte TX buffer, 48 byte object... 
Wonder to they all need to be that large.  Example on Teensy, the defaults for Serial objects
Serial1(TX=64, RX=64), Serial2(40, 64), Serial3(40, 64), so for example if we reduce their sizes to 64 and 64, again we save (256*3) data bytes... 

Of course there are more radical changes, where hopefully you can setup the code, such that if a sketch does not use any of some main feature, than the code and data objects associated with it are not brought into the binary.   Probably more than I want to chew ;)

OpusK, 2018-11-19:
Changing the buffer size or building only the necessary Serials (including SerialEvent processing) can affect existing examples. 
However, this is a policy issue, so internal discussion is necessary.

First, regarding the buffer size, I think we can reduce the size if there is no problem with DXL communication.

Second, as far as I can see, Arduino's official boards have created all of their Serial classes globally, so we seem to have done so too. If we need this, it will be enough to add developer options.

OpusK, 2018-11-19:
@KurtE ,

Thank you for your good suggestion.
With serialEvent() it will be a solution, but it will be difficult to apply because there are Serial1,2,3 classes that are used in existing TB3 and other examples.

However, it is a good idea to reduce the default buffer size of the Serial class (about 64 bytes)
Of course, the DXL port still needs a large buffer, so this should be fixed. Also, in the case of the LDS example, a large buffer is required because the amount of LDS data is large. To solve this problem, we can add an API that can change the pointer to a buffer created by the user.

For EEPROM, it would be better to change it to const.

What do you think?

OpusK, 2019-06-12:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

KurtE, 2019-06-12:
@OpusK - makes sense to close this one.  

This can be an ongoing type of thing, that every so often someone should take a look at the sizes of things that are part of the system and see if there are any low hanging fruit. 

Examples of Serial buffers and the like.    With only 3 serial objects, maybe not as critical, but for example some Teensy boards have up to 8 serial devices... An in their case for the T4, I did do some stuff similar to mentioned above.  The SerialX objects each go into their own source file, along with their own buffers and their own default SerialEventX function.  

The main loop code that did stuff like if (Serial2.available()) SerialEvent2();
code instead ran off a function pointer array, and if slot(2) had a function pointer would it get called...
And the default SerialEvent2 object removes itself from the list...  SerialX.begin() adds itself to the list.  So only if someone calls Serial2.begin() will the code for Serial2.available() be called and if the user did not provide their own copy of SerialEvent2 to overwrite the weak version, then the first time some character comes in the code to call this, it will disable this checking...  Not perfect, as someone could do a Serial2.begin(), and maybe actually never uses Serial2, so the main loop will continue to call Serial2.available()...

OpusK, 2019-06-12:
@KurtE 

This issue reopened.

Currently, I have other things to deal with, so after I finish them first, I will look back at this issue.
Thank you for your contribution :)

KurtE, 2024-08-16:
Sort of old and who knows what applies any more so, going to go ahead and close it out

closed 2024-08-16 by KurtE