ROBOTIS-GIT/OpenCM9.04 #60: EEPROM Writes - Too many EEPROM.write() can crash board - needs recovery...

KurtE, 2018-11-20:
While testing, my memory reduction code, with EEPROMs, I thought I should try to verify that EEPROM still worked.  So I tried a simple program to write to all memory locations and then try to read them back.  When I ran it, it hung and the board become totally unresponsive.  If you reset the board, it will not run.  You need to do a firmware recovery to reuse the board.  

I tested this on the released version of OpenCM9.04 Arduino release and it had the same problem...  Note: if I have the test case only output 64 items in the run it appears to work, but if you reboot and try again, reboot... on the 4th run it will die the same way....  Right now I have it configured to try to output 256 items and it does it in the first run.
```
#define MAX_INDEX 256
#include <EEPROM.h>
void setup() {
  // put your setup code here, to run once:
  while (!Serial) ;
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(115200);
  Serial.println("EEPROM Test");
  // lets try writing to EEPROM
  uint16_t max_index = EEPROM.length();
  if (max_index > MAX_INDEX) max_index = MAX_INDEX;
  Serial.printf("EEPROM length: %d Max Test: %d\n", EEPROM.length(), max_index);
  uint32_t start_time = micros();
  for (uint16_t i = 0; i < max_index; i++) {
    EEPROM.write(i, i & 0xff);
  }
  Serial.printf("Write time: %d\n", micros() - start_time);
  Serial.println("Start read back");
  start_time = micros();
  for (uint16_t i = 0; i < max_index; i++) {
    uint8_t b = EEPROM.read(i);
    if (b != (i & 0xff)) {
      Serial.printf("Read mismatch: %d %x!=%x\n", i, b, i & 0xff);
    }
  }
  Serial.printf("Read time: %d\n", micros() - start_time);
}

void loop() {
  digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
  delay(250);
 }
```

I then instrumented the EEPROM code in drv_eeprom.c 
```
static uint16_t EE_VerifyPageFullWriteVariable(uint16_t VirtAddress, uint16_t Data)
{
  HAL_StatusTypeDef flashstatus = HAL_OK;
  uint16_t validpage = PAGE0;
  uint32_t address = EEPROM_START_ADDRESS, pageendaddress = EEPROM_START_ADDRESS+PAGE_SIZE;
  vcp_printf("EEVPFWV %x %x ", VirtAddress, Data); vcp_flush_tx();;

  /* Get valid Page for write operation */
  validpage = EE_FindValidPage(WRITE_IN_VALID_PAGE);

  /* Check if there is no valid page */
  if (validpage == NO_VALID_PAGE)
  {
    vcp_printf("NO_VALID_PAGE\n"); vcp_flush_tx();
    return  NO_VALID_PAGE;
  }

  /* Get the valid Page start address */
  address = (uint32_t)(EEPROM_START_ADDRESS + (uint32_t)(validpage * PAGE_SIZE));

  /* Get the valid Page end address */
  pageendaddress = (uint32_t)((EEPROM_START_ADDRESS - 1) + (uint32_t)((validpage + 1) * PAGE_SIZE));

  vcp_printf(" PAGE%d %x %x ", (validpage==PAGE0)? 0 : 1 , address, pageendaddress); vcp_flush_tx();
  /* Check each active page address starting from begining */
  while (address < pageendaddress)
  {
    /* Verify if address and address+2 contents are 0xFFFFFFFF */
    if ((*(__IO uint32_t*)address) == 0xFFFFFFFF)
    {
      /* Set variable data */
      vcp_printf(" HFP %x %x %x  ", FLASH_TYPEPROGRAM_HALFWORD, address, Data); vcp_flush_tx();
      flashstatus = HAL_FLASH_Program(FLASH_TYPEPROGRAM_HALFWORD, address, Data);
      /* If program operation was failed, a Flash error code is returned */
      if (flashstatus != HAL_OK)
      {
        vcp_printf(" fail %x\n", flashstatus); vcp_flush_tx();

        return flashstatus;
      }
      /* Set variable virtual address */
      vcp_printf(" HFP %x %x %x  ", FLASH_TYPEPROGRAM_HALFWORD, address + 2, VirtAddress); vcp_flush_tx();
      flashstatus = HAL_FLASH_Program(FLASH_TYPEPROGRAM_HALFWORD, address + 2, VirtAddress);
      /* Return program operation status */
      vcp_printf(" status %x\n", flashstatus); vcp_flush_tx();
      return flashstatus;
    }
    else
    {
      /* Next address location */
      address = address + 4;
    }
  }

  /* Return PAGE_FULL in case the valid page is full */
  vcp_printf("PAGE_FULL\n"); vcp_flush_tx();
  return PAGE_FULL;
}
```
I recovered the firmware and ran the test again, with this stuff and it spewed the data...
```
EEPROM Test
EEPROM length: 512 Max Test: 256
EEVPFWV 0 0  PAGE0 801f800 801fbff  HFP 1 801f804 0   HFP 1 801f806 0   status 0
EEVPFWV 1 1  PAGE0 801f800 801fbff  HFP 1 801f808 1   HFP 1 801f80a 1   status 0
EEVPFWV 2 2  PAGE0 801f800 801fbff  HFP 1 801f80c 2   HFP 1 801f80e 2   status 0
EEVPFWV 3 3  PAGE0 801f800 801fbff  HFP 1 801f810 3   HFP 1 801f812 3   status 0
EEVPFWV 4 4  PAGE0 801f800 801fbff  HFP 1 801f814 4   HFP 1 801f816 4   status 0
EEVPFWV 5 5  PAGE0 801f800 801fbff  HFP 1 801f818 5   HFP 1 801f81a 5   status 0
EEVPFWV 6 6  PAGE0 801f800 801fbff  HFP 1 801f81c 6   HFP 1 801f81e 6   status 0
EEVPFWV 7 7  PAGE0 801f800 801fbff  HFP 1 801f820 7   HFP 1 801f822 7   status 0
EEVPFWV 8 8  PAGE0 801f800 801fbff  HFP 1 801f824 8   HFP 1 801f826 8   status 0
EEVPFWV 9 9  PAGE0 801f800 801fbff  HFP 1 801f828 9   HFP 1 801f82a 9   status 0
EEVPFWV a a  PAGE0 801f800 801fbff  HFP 1 801f82c a   HFP 1 801f82e a   status 0
EEVPFWV b b  PAGE0 801f800 801fbff  HFP 1 801f830 b   HFP 1 801f832 b   status 0
EEVPFWV c c  PAGE0 801f800 801fbff  HFP 1 801f834 c   HFP 1 801f836 c   status 0
EEVPFWV d d  PAGE0 801f800 801fbff  HFP 1 801f838 d   HFP 1 801f83a d   status 0
EEVPFWV e e  PAGE0 801f800 801fbff  HFP 1 801f83c e   HFP 1 801f83e e   status 0
EEVPFWV f f  PAGE0 801f800 801fbff  HFP 1 801f840 f   HFP 1 801f842 f   status 0
EEVPFWV 10 10  PAGE0 801f800 801fbff  HFP 1 801f844 10   HFP 1 801f846 10   status 0
...
EEVPFWV f7 f7  PAGE0 801f800 801fbff  HFP 1 801fbe0 f7   HFP 1 801fbe2 f7   status 0
EEVPFWV f8 f8  PAGE0 801f800 801fbff  HFP 1 801fbe4 f8   HFP 1 801fbe6 f8   status 0
EEVPFWV f9 f9  PAGE0 801f800 801fbff  HFP 1 801fbe8 f9   HFP 1 801fbea f9   status 0
EEVPFWV fa fa  PAGE0 801f800 801fbff  HFP 1 801fbec fa   HFP 1 801fbee fa   status 0
EEVPFWV fb fb  PAGE0 801f800 801fbff  HFP 1 801fbf0 fb   HFP 1 801fbf2 fb   status 0
EEVPFWV fc fc  PAGE0 801f800 801fbff  HFP 1 801fbf4 fc   HFP 1 801fbf6 fc   status 0
EEVPFWV fd fd  PAGE0 801f800 801fbff  HFP 1 801fbf8 fd   HFP 1 801fbfa fd   status 0
EEVPFWV fe fe  PAGE0 801f800 801fbff  HFP 1 801fbfc fe   HFP 1 801fbfe fe   status 0
EEVPFWV ff ff  PAGE0 801f800 801fbff PAGE_FULL
EEVPFWV ff ff  PAGE1 802f800 802fbff 
```
So it died/hung when it transitioned from Page 0 to Page 1.   Wonder if Page 1 is valid?

Time to debug...

KurtE, 2018-11-20:
Looking at this, what I find interesting is that: 
In drv_eeprom.c lines 126...

#define ADDR_FLASH_PAGE_126   ((uint32_t)0x0801F800) /* Base @ of Page 126, 1 Kbytes */
#define ADDR_FLASH_PAGE_127   ((uint32_t)0x0801FC00) /* Base @ of Page 127, 1 Kbytes */

So this page looks correct for PAGE0:
EEVPFWV 0 0  PAGE0 801f800 801fbff  HFP 1 801f804 0   HFP 1 801f806 0   status 0

But for Logical Page1, which are first write is showing: EEVPFWV ff ff  PAGE1 802f800 802fbff 
Which is not Page_127...

The function EE_FindValidPage returns PAGE0 or PAGE1
Which are defined: 
```
#define PAGE0                 ((uint16_t)0x0000)
#define PAGE1                 ((uint16_t)0x0040)
```
The computation for The pages address is: 
```
 address = (uint32_t)(EEPROM_START_ADDRESS + (uint32_t)(validpage * PAGE_SIZE));
pageendaddress = (uint32_t)((EEPROM_START_ADDRESS - 1) + (uint32_t)((validpage + 1) * PAGE_SIZE));
```

So wondering why PAGE1 is defined as 0x0040?  Wonder what would happen if it was defined as 0x01?

OpusK, 2018-11-20:
@KurtE ,

> So wondering why PAGE1 is defined as 0x0040? Wonder what would happen if it was defined as 0x01?

Oh, sorry and thanks...
The value of PAGE1 is invalid. The problem does not occur until 0x01.
If I make a correction, it will affect your PR.
Do you want to modify and apply PR?

KurtE, 2018-11-20:
@OpusK 
That I can do...

OpusK, 2018-11-20:
@KurtE,

Thanks again for your contribution!

closed 2018-11-26 by OpusK