ROBOTIS-GIT/OpenCM9.04 #63: BT-410 & OpenCM9.04 via Arduino IDE

GrigoriyAY, 2018-11-28:
Hello,

I'm interesting in setting of wireless communication between Arduino IDE (Windows 10) and OpenCM9.04 via Bluetooth module BT-410. Could you help me how I can carry out this communication?

GrigoriyAY, 2018-11-29:
@OpusK ,

It's not clear. I used this manual and manual for BT-410. But how can I get information from OpenCM in port monitor (Arduino IDE) or using another serial communication (in MATLAB for example), if I don't know the number of serial port of BT-410? The module is displayed in the device manager not as a COM port.

OpusK, 2018-11-29:
@GrigoriyAY ,

Did you connect [BT-410 dongle](http://emanual.robotis.com/docs/en/parts/communication/bt-410-dongle/) to PC?

More information is needed. Including your connection diagram and device manager screenshots.

GrigoriyAY, 2018-11-29:
@OpusK ,

Clearly, I have already begun to assume that I need BT-410 dongle. That is, connecting only BT-410 to OpenCM, I will not be able to communicate with the computer? Earlier I used to work only with Arduino and bluetooth module HC-06 and there was no need to connect an additional module to the computer.  Are there any other solutions, I really would not want to take a USB-port of a laptop by something (there are not so much).

OpusK, 2018-11-29:
@GrigoriyAY ,

Unfortunately, the BT-410(BLE) is not compatible with regular Bluetooth dongles.
Typically, Classic Bluetooth (Bluetooth 2.0) has compatible dongles, but BLEs vary from vendor to vendor.

So if you do not have a BT-410 dongle, you will not be able to communicate with your PC.

OpusK, 2018-11-29:
@GrigoriyAY 

Yes, it is possible.
Please refer to [E-manual](http://emanual.robotis.com/docs/en/parts/communication/bt-210/) for more details.

GrigoriyAY, 2018-11-29:
@OpusK , thank you.

closed 2018-11-29 by OpusK

GrigoriyAY, 2019-06-23:
@OpusK Hello, I have new question about bluetooth connection. Is it possible to connect BT-410 via LN-101 to PC and communicate with BT-410 connected to OpenCM9.04? At this time I checked setting of BT-410 for OpenCM, but I couldn't set communication with BT-410 connected to PC via LN101 (although, I read the possibility of this in old manual: https://fccid.io/SOD-BT-410/User-Manual/User-Manual-2537546)

ROBOTIS-Shibata, 2021-03-12:
@Y-Grigoriy 
Hi, LN-101's TX pin and RX pin are the opposite of OpenCM 9.04's UART port. So you need to reverse TX and RX of BT connection cable. In other words, you need to make a crossover cable. 
**Note :  this method is not officially supported by ROBOTIS**. 
[image]

Y-Grigoriy, 2021-03-31:
@ROBOTIS-Shibata, thank you for this trick)