ROBOTIS-GIT/OpenCM9.04 #69: st-link simple example

Le0nX, 2019-02-08:
Hello! I'm new to ARM programming and couldn't find any manuals/examples on C++ programming vis ST-Link for your board... Could you give some simple example of code and process of it's building with flashing OpenCM via ST-link? 

For example, I have Dynamixel ax-12a servos and want to rotate one of them with C++ bare bones on your board. Will I be able to use Dynamixel SDK for this task?

Le0nX, 2019-02-12:
anybody?

OpusK, 2019-02-13:
Hi, @Le0nX 

We do not officially provide a reference for this method.
However, since there is a hardware SWD pin, you can use it to create general firmware yourself.

If you use Eclipse or use TrueStudio, you can do this easily.
However, in this case, OpenCM's boot loader will be erased, which can be a bit cumbersome to repair.

TuxLeon, 2019-02-20:
Hi Le0nX,
I was also looking for the STM source code or project. I now found this github repository. Perhaps this is what you were looking for.
https://github.com/ROBOTIS-GIT/OpenCM9.04/tree/master/arduino/opencm_arduino/opencm9.04

OpusK, 2019-02-21:
Hi, @TuxLeon.

This is a project to create a .lib file for opencm9.04. With this eclipse project, you can not do exactly what LeOnX wants. Of course, you can refer to it to create an eclipse or TruStudio project.

closed 2019-02-28 by Le0nX