ROBOTIS-GIT/OpenCM9.04 #70: Unable to upload code on OpenCM 9.04

tr-gis, 2019-02-18:
I am trying to upload one of the examples (Examples -> OpenCM 9.04 -> 01_Basics -> b_Blink_LED) on the OpenCM 9.04 Type B board. Initially when I connect the board using micro-usb cable, only the Power LED (Red color) lights up. When I try to upload the example I get the following error:

```
Sketch uses 34404 bytes (29%) of program storage space. Maximum is 116736 bytes.
Global variables use 8644 bytes of dynamic memory.
stm32ld ver 1.0.1
Unable to connect to bootloader
OpenCM Download Ver 1.0.4 2015.06.16 
Fail to connect OpenCM
```

Next I connected the Board when pressing down on User SW button. This time both Power (Red) and Status (Green) LEDs light up. Now if I try to upload the example code I get the following error:

```
Sketch uses 34404 bytes (29%) of program storage space. Maximum is 116736 bytes.
Global variables use 8644 bytes of dynamic memory.
Enter bootloader
```

Could you please point out how to solve the issue.

OpusK, 2019-02-19:
Hi, @tr-gis 

Could you tell me your OS information?
The information you gave us the second time is correct. 
However, it is estimated that there is a problem with data transmission.
If possible, try changing the USB cable.

tr-gis, 2019-02-19:
Hi @OpusK ,

I am using Ubuntu 16.04. I tried with other USB cables just like you suggested but still get the same `Enter bootloader` message. I have assigned 777 permission to the /dev/ttyACM0 port (to which the board is connected.) 

What should be the selected under Tools -> Programmer ?. Currently it is set to AVRISP mkII.

OpusK, 2019-02-20:
> What should be the selected under Tools -> Programmer ?. Currently it is set to AVRISP mkII.

This setting should be set to dfu-util when you are booting the boot loader. It does not matter in this situation.

And did you add the rule file through the document below?
- [usb port setting](http://emanual.robotis.com/docs/en/parts/controller/opencm904/#usb-port-settings)

tr-gis, 2019-02-21:
Yes. I added the usb rule file.

OpusK, 2019-02-21:
Hi, @tr-gis 
Do you have any other PC that you can test? I can not reproduce the situation like you.

tr-gis, 2019-02-21:
Ok, I will try it out from another PC and get back. Thanks :)

gpwhd07, 2019-03-14:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2019-03-14 by gpwhd07

ruffson, 2019-05-11:
Hi I have the same problem.
My setup is 4 `XL430-W250-T` daisy-chain-connected to a `OpenCM 485 Expansion Board` which is connected to the `OpenCM 9.04`. I could execute scripts fine before, both via the "old" API as well as using the newer Workbench API.

But now any code I try to upload via the Arduino IDE results in this output:
(Example here is the `b_Blink_LED`)
```
Sketch uses 34404 bytes (29%) of program storage space. Maximum is 116736 bytes.
Global variables use 8644 bytes of dynamic memory.
Unable to connect to bootloader
stm32ld ver 1.0.1
OpenCM Download Ver 1.0.4 2015.06.16 
Fail to connect OpenCM
```

I am running macOS 10.14.4, tried different USB ports, rebooted my computer and shut of the board. Nothing seems to make this work anymore. Is there a way to reset the board to get to a working state again?

OpusK, 2019-05-13:
Hi, @ruffson 

> After trying several combinations of pressing (and keeping) the two buttons on the board (USER SW and RESET) it seems to have recovered. Although this does not seem to be covered in any emanual.

This is covered in the e-manual. Please see the link below.
- [recovery mode](http://emanual.robotis.com/docs/en/parts/controller/opencm904/#recovery-mode)

Did you mean that the recovery mode method solves this problem?

ruffson, 2019-05-17:
> Hi, @ruffson
>
>
> This is covered in the e-manual. Please see the link below.
>
>     * [recovery mode](http://emanual.robotis.com/docs/en/parts/controller/opencm904/#recovery-mode)
>
>
> Did you mean that [...]

Thank you, that worked!