ROBOTIS-GIT/OpenCM9.04 #73: SerialX - Half duplex support - Support FlySky IBUS

KurtE, 2019-04-01:
Not sure if this is the best place to ask and or discuss... 

**Background:**
I am playing with a prototype new version of the Trossen Robotics PhantomX hexapod that is now built with XL430-W250 servos instead of AX-12/18 servos.  Also they would like to get away from using their on Arbotix-M board (Atmega644p) and use the OpenCM board and maybe get away from their Arbotix Commander remote (XBEE)... More details of current PhantomX is up at: 
https://www.trossenrobotics.com/phantomx-ax-hexapod.aspx .  So looking at some alternatives.  I personally want the remote to have a at least one or two analog joysticks..  

With some other processors I would probably go with PS3 or PS4 like remotes but not sure how to use them here... 

So I am now experimenting with a Fly Sky Remote (FS-i6S) with an FS-IA6B receiver as they are reasonably cheap and they have the ability to talk to a PC using TTL Serial at 115200.   I have now been able to rig up a connection to Serial2 on OpenCM board to receive 10 channels of data from the remote.  Another interesting thing about this remote is there is some ability for the receiver to send back data that can be displayed on the remote.   They do this using a half duplex Serial connection.  Some of this is talked about at: https://github.com/betaflight/betaflight/wiki/Single-wire-FlySky-(IBus)-telemetry

**Experimenting**
Currently I have a test program that does some of this currently using two Serial ports.  The Serial2 to receive the SERVO data and the bidirectional SENS(or) code that I have hooked up to a DXL TTL pin on OpenCM board and hacked up code to use the DXL code to control direction.

But I was thinking of experimenting with trying to do both using Serial2... 

**Question:**
So the question is, has anyone tried using the processors hardware support for half duplex? 
Some of this is decribed in section 27.3.10 of the STM32F10x Reference manual. 

I know that the HAL layer has some support for this, in the function; HAL_HalfDuplex_Init
But I don't think anyone calls it.  Was wondering if it might make sense to try adding a new UARTMode like Mode_8N1_HALFDUPLEX which could be passed in through 
Serial2.begin(115200, Mode_8N1_HALFDUPLEX )

The underlying code would then have to pick off this flag and call off to HAL_HalfDuplex_Init instead of HAL_UART_Init.  My guess is it would also require additional changes as the initialization of the IO pins would probably be different.  RX Pin not used TX pin supposed to be configured either floating input (or output high open-drain)

From description could not tell if that pin configuration for TX would be done at init time and/or when we detect that there is no more data to output.

So again I am wondering if anyone has tried something like this?  Make sense?  Or better to just punt for now? 
Thanks
Kurt

OpusK, 2019-04-02:
Hi, @KurtE 

Long time no see! :)

> So the question is, has anyone tried using the processors hardware support for half duplex?

I have never tried this.

> But I don't think anyone calls it. Was wondering if it might make sense to try adding a new UARTMode like Mode_8N1_HALFDUPLEX which could be passed in through
Serial2.begin(115200, Mode_8N1_HALFDUPLEX )

With respect to #72, we are considering the API you mentioned. Maybe we should discuss how to get the driver structure associated with Halfduplex while applying this API.

KurtE, 2019-04-02:
Hi @OpusK, 

I have been around, trying to give some answers up on your forum, but been spending most of my time playing with the Beta of Teensy 4 (https://forum.pjrc.com/threads/54711-Teensy-4-0-First-Beta-Test)  I have a beta1 board and hope to get a beta 2 board at some point.

As for #72 I looked at it and a few days ago and thought about commenting.  The hard part with that one is going to 8E2 implies 9 bit data, which the Serial device can support,  but it then implies that both the RX and TX buffers need to change from 8 bit to 16 bit.  And that needs to trickle through everything, like the DMA communications and the like.   On a Teensy, the user has to edit the source files to enable 9 bit mode, which then all buffers are setup with 16 bit buffers...   So might be a bit messy. 

I am hoping that half duplex will not be as messy. 

Also API wise, currently you have:
```
   enum UARTModes {
      Mode_8N1 = 0, // = US_MR_CHRL_8_BIT | US_MR_NBSTOP_1_BIT | UART_MR_PAR_NO,
      Mode_8E1,     // = US_MR_CHRL_8_BIT | US_MR_NBSTOP_1_BIT | UART_MR_PAR_EVEN,
      Mode_8O1,     // = US_MR_CHRL_8_BIT | US_MR_NBSTOP_1_BIT | UART_MR_PAR_ODD,
      Mode_8M1,     // = US_MR_CHRL_8_BIT | US_MR_NBSTOP_1_BIT | UART_MR_PAR_MARK,
      Mode_8S1      // = US_MR_CHRL_8_BIT | US_MR_NBSTOP_1_BIT | UART_MR_PAR_SPACE,
    };
...
     void begin(const uint32_t dwBaudRate, const UARTModes config);
```
Which is fine, but I know on some platforms, they don't fix the begin method to only the enum values, but instead, setup something like a 16 bit or 32 bit value for config, where then you can encode more information into the actual value.  For example maybe the lower 3 or 4 bits is the actual encoding (none, Even, Odd, Mark, Space)  Maybe another bit specifies 1 stop or 2 stop.  

On Teensy there are some other bits/configuration options which I have not looked into here, where include: 
a) Invert the RX data
b) invert the TX data
Maybe some others on some of the boards, like 1.5 stop bits... 

Over the next few days, I may try hacking in half duplex and see how bad it is.

KurtE, 2019-04-06:
@OpusK, 

I started playing.  

Step 1) was to setup to pass through the stuff given in the config parameter of SerialX.begin() to the hal layer.   Note: I reorganized some of these and dropped some: 
``` 
public:
    enum UARTModes {
      Mode_8N1 = 0,         // No special options
      Mode_7E1 = 0x02,      // CR1 PCE=1, PS=0
      Mode_7O1 = 0x03,      // CR1 PCE=1, PS=1
      Mode_8N1_HALF=0x80,   // 8N1 but supports Half duplex. 
      Mode_8N2 = 0x20,      // CR2 STOP=2
      Mode_7E2 = 0x22,      // CR1 PCE=1, PS=0, CR2 STOP=2
      Mode_7O2 = 0x23,      // CR1 PCE=1, PS=1, CR2 STOP=2
      Mode_8N2_HALF=0xA0,   // 8N2 but supports Half duplex. 

//      Mode_8M1,     // = US_MR_CHRL_8_BIT | US_MR_NBSTOP_1_BIT | UART_MR_PAR_MARK,
//      Mode_8S1,      // = US_MR_CHRL_8_BIT | US_MR_NBSTOP_1_BIT | UART_MR_PAR_SPACE,
    };
    // Setup the Modes to be bit settings: Bits LSB=0, MSB=7
    // 0-1 Parity (CR1)
    // 4-5 Stop bits (CR2 only defined 1=0, 2=2, but .5 and 1.5 are valid as well... )
    // 8 - Half duplex mode (CR3)
```

Actually wondering if I should redo these names and the location of these names.  That is if you look at the Arduino documentation: https://www.arduino.cc/en/serial/begin

You will see names like:
```
SERIAL_8N1 (the default)
SERIAL_8N2
SERIAL_7E1 
SERIAL_7O1 
```
Where currently with this code you have to do something like:
```
Serial2.begin(115200, UARTClass::Mode_7E2);
```
Note: I did try running simple sketch: 
```
void setup() {
  while (!Serial) ; // wait for USB to be connected
  Serial.begin(115200);
  Serial2.begin(115200, UARTClass::Mode_7E2);
}

void loop() {
  while (Serial.available()) {
    Serial2.write(Serial.read());  
  }
  while (Serial2.available()) {
    Serial.write(Serial2.read());
  }

}
```
Where I would change the begin parameter and then look at the output to Logic Analyzer and I think I have 8N1,7E1,7O1, Plus I tried 7N2, 7O2 and I think have them working.

Current WIP Fork/Branch is up at: https://github.com/KurtE/OpenCM9.04/tree/Serial_config

Note: I have not tried yet to see how hard it would be to enable 8O2 or 8E2 (or the like) as I mentioned  in #72 these would require the data paths to converted from 8 bit data to 9 (actually 16 bit) data... 

Next up to see how hard it would be to use the Duplex settings...

OpusK, 2019-04-09:
@KurtE 

As you say, there is a lot of need to worry about 9bit.
Therefore, I think that it is better to implement the processing of data within 8 bits first.

I saw the work going on in your branch.
It is awesome!!
I would like to PR your branch and review it.

KurtE, 2019-04-10:
Still WIP - as for the half duplex stuff.  I have some code in place on my machine which I am trying to debug (fix)

Current checked in stuff does not have all of the release header files updates nor lib file.

Also still wondering if we should put in the names for the different Open modes like is shown in Arduino help files instead of member:enum name... 

Let me know if you wish to take in the current WIP while I am debugging the rest.
Current stuff put in PR: #

KurtE, 2024-08-16:
No idea if any of this still applies...

closed 2024-08-16 by KurtE