ROBOTIS-GIT/OpenCM9.04 #76: Can't unlock the stm32f103

Le0nX, 2019-04-26:
Hi! When I'm trying to flash the stm32f103 on the opencm 9.04 board via st-link, openocd gets an error:
```
Info : device id = 0x20036410
Info : flash size = 128kbytes
Error: stm32x device protected
Error: failed erasing sectors 0 to 2
```
I've tried `stm32f1x unlock 0` but it didn't help...

Le0nX, 2019-04-26:
[image]

Le0nX, 2019-04-27:
I'm using STLINK-V2 & SW4STM32 IDE + openocd

Le0nX, 2019-04-29:
I've used stm utils for windows and it has solved the problem. After it had unlocked and erased the chip you can flash it under linux...

Le0nX, 2019-04-29:
Still don't understand why openocd doesn't want to unlock some of opecm's mem sectors...

TuxLeon, 2019-04-29:
Could it be related to the version of your ST-Link Utility like in this
post?
https://community.st.com/s/question/0D50X00009XkWyMSAV/stlink-utility-mass-flash-erase-fails-however-erase-bank-1-and-2-works

Le0nX, 2019-04-29:
@TuxLeon, do you mean ST-Link Utility under linux? 
ST-Link Utility for Windows does it's job very well: unlocks and erases/flashes the chip.
The problem was with linux version. Under linux I couldn't unlock and erase the chip. Now, after I've done this under windows, I can flash the unlocked chip under linux.

TuxLeon, 2019-04-29:
Yes. Perhaps just try an other version of openOcd under Linux.

closed 2019-05-20 by Le0nX