ROBOTIS-GIT/OpenCM9.04 #77: Deepsleep

olsky, 2019-05-05:
Hi there,

could you please guide on how to use the deepSleep functionality present in the stm32? 

just like from: https://github.com/stm32duino/STM32LowPower

thank you

OpusK, 2019-05-08:
@olsky 

Currently, it is not implemented.
However, I will discuss the addition of related features with the relevant team.

olsky, 2019-05-09:
@OpusK 

thank you for your message. Could you please estimate, is it any big issue or in relatively short time feasible...

Is there any workaround in the meantime? 

What I also have difficulties with is: Timer / Alarm / RTC ... are there any examples? 

Thanks!

OpusK, 2019-06-13:
@olsky 

I'm sorry for the late reply.

>Could you please estimate, is it any big issue or in relatively short time feasible...

Unless they are needed at the Arduino Core API level, including "DeepSleep", I think it will be difficult for the company to proceed with development. Also, it seems that it is difficult to add functions soon as other schedules are currently being held.

OpenCM uses ST chip, so if you use HAL Library directly, you can use Timer, RTC, etc. to your desired level.

closed 2019-07-17 by OpusK