ROBOTIS-GIT/OpenCM9.04 #80: Using OpenCM9.04 + Exp 485 boards to communicate directly with ROS

Iervolino458, 2019-05-17:
Hello,

I'm currently having problems to communicate the OpenCM9.04 + EXP 485 boards with the dynamixel_workbench ROS package. I can detect the motors fine using the find_dynamixel script with Arduino IDE, but the ROS package node can't find any motors.

I tried to use the script below to send communication from Serial (USB) over Serial3 (OpenCM 485) but ROS package still hasn't detected any motors:

`#define CMD_PORT Serial   // USB
#define DXL_PORT Serial3  // UART1
#define DXL_BAUD 1000000

void setup() {
  CMD_PORT.begin(115200);
  DXL_PORT.begin(DXL_BAUD);
}

void loop() {
  update_dxl();
}

void update_dxl() {
  // USB TO UART1
  if (CMD_PORT.available()) {
    DXL_PORT.print((char) CMD_PORT.read());
  }

  // UART1 to USB
  if (DXL_PORT.available()) {
    // toggleLED();
    CMD_PORT.print((char) DXL_PORT.read());
  }
}`

Is there any usb_to_dxl script that I can use with the OpenCM9.04 + EXP 485 boards to communicate directly with ROS?

Thanks.

OpusK, 2019-06-13:
Hi, @Iervolino458 

OpenCM does not provide a bypass example separately.

There is a problem with your code. Because DYNAMIXEL is half-duplex, TX and RX must be determined with the DIR pin. Your code does not have this capability, so it seems that communication is not possible.

I think it would be helpful to refer to the [usb_to_dxl example in OpenCR](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/OpenCR/examples/10.%20Etc/usb_to_dxl/usb_to_dxl.ino).

This is just a reference code and does not guarantee behavior.

Iervolino458, 2019-06-13:
@OpusK

I fixed the issue myself using DynamixelSDK to communicate data between Serial and Serial3.

I'm having another problem, however, regarding the motors velocity. I'm using XM540-W270 motors and I can't seem to slow velocity down even when setting velocity and acceleration parameters (control table). The motor is in position mode by the way.

How can I limit the motors velocity keeping the position operation mode?

OpusK, 2019-06-13:
@Iervolino458 

You said you are using the workbench ROS package, are you using the SDK package now?
Anyway, it would be nice to raise the issue for each reporter issue.
- [Workbench](https://github.com/ROBOTIS-GIT/dynamixel-workbench)
- [SDK](https://github.com/ROBOTIS-GIT/DynamixelSDK)

Iervolino458, 2019-06-13:
@OpusK

I'm using DynamixelWorkbench package to interface the motors with ROS. I've used SDK just as the usb port handler to develop the usb tosser.

closed 2019-06-13 by Iervolino458