ROBOTIS-GIT/OpenCM9.04 #81: barebones USART3 corrupts data at 1mbit

Le0nX, 2019-05-20:
I've got an issue with comunicating with AX-12a trough USART3. When a baud rate is 115200 get the right data, but when I change the baud to 1mbit(for ax-12a) then I get corrupted data...

Le0nX, 2019-05-20:
If I want to send data to EXP 485 board from opencm 9.04 and my ax-12a is connected to this EXP board, do I need to do it via USART1 or USART3?

Le0nX, 2019-05-25:
I've fixed the problem. Everything is working great with FreeRTOS running on Opencm 9.04 with EXP board connected.

closed 2019-05-25 by Le0nX

Iervolino458, 2019-05-25:
Hello, how did you manage to communicate the motors connected in the EXP board to the OpenCM usb?

Le0nX, 2019-05-26:
Hi, @Iervolino458! Nope. The ax-12a motors are connected to EXP board via UART(TTL), not via USB. So I'm sending commands to servos from Opencm 9.04 connected to EXP board trough USART3.