ROBOTIS-GIT/OpenCM9.04 #85: Communication problem with OpenCM9.04 via ArduinoIDE

mkdy, 2019-07-10:
Hello,

I've installed ArduinoIDE on my Ubuntu laptop and connected OpenCM9.04 board to it via USB. So, I'm able to run BlinkLED sketch and green LED really blinks. However, when I'm trying to run Serial_HelloWorld sketch, I see it is uploaded and says "Go Application", but I find neither counts nor any errors in terminal.

Any ideas why that happens?

jawad-tech, 2019-07-15:
Have you opened you SERIAL monitor after the program is uploaded successfully. Because When i execute the Position control, the motor doesn't rotate, but When i open the SERIAL monitor after the program is uploaded, the motors start to rotate.

OpusK, 2019-07-17:
Hi, @mkdy 

The above terminal logs are normal.
Serial example is to output data in Serial, so check with Serial Monitor.

It is normal for the Arduino IDE terminal not to output any other data.

mkdy, 2019-07-20:
@jawad-tech @OpusK Thanks for the responses!
Indeed, when I open Serial Monitor, I see Hello-World messages. However, I have to change port from /dev/ttyACM0 to /dev/ttyACM1 manually before opening Serial Monitor (otherwise Serial Monitor would not open, and that's weird).
At the moment I'm trying to start Dynamixel XL430 servo, but with no success. FindDynamixels sketch says that no devices are found. XL430 is powered via 12v5a power supply. Any ideas?

jawad-tech, 2019-07-21:
1) First When you connect the usb from Controller to pc, you will check in ubuntu terminal 'lsusb` command to see your Controller IS connected to Which port. because Sometime there is a bug in an arduino port. 
also try to make the port permissions by
'sudo chmod 777 /dev/ttyACMx  Where x is your connected port.

and one other thing If you are using arduino IDE to Run the Code Its Ok, 
Please check the SERIAL settings
+ If using OPENCM Controller SERIAL bus1
+ If using OPENCM and rs485 board then serIal  bus3 
+If open cr then SERIAL bus2

check your Connections of dynamixel to Controller or U2D2

OpusK, 2019-07-30:
@mkdy 

Did you upload or reset the firmware with Serial Monitor open?
If so, it can be changed to a different name (eg. ACM1) by avoiding the device name already in use.

When uploading a reset or a sketch file, close the serial monitor window.

closed 2019-08-29 by OpusK