ROBOTIS-GIT/OpenCM9.04 #86: OpenCM 9.04 problem with linux and ROS

jawad-tech, 2019-07-15:
Hello, I am using OpenCM9.04 for my project. I have passed through different difficulties.

1) PC1 with AMD64 architecture works fine with Robotis when I controlled with Windows 10. The motors aka dynamixels are pinged successfully and work fine.

2) I have downloaded and follow all the tutorials to build by using ROS and linux but when I ping the dynamixels so it dont find any dynamixels. I have also tried position control, but it gives error TxxRxx: The status packet is missing or some error about status packet 

2) PC2 with aarch64 architecture: (Using Linux)
 2.1) I have downloaded and follow all the tutorials to build by using ROS and linux but when I ping the dynamixels so it dont find any dynamixels. I have also tried position control, but it gives error TxxRxx: The status packet is missing or some error about status packet 

2.2) I then used Arduino IDE to code OpenCm9.04 but when I port to arduino and installing OPENCM9.04 library it gives "opencm-gcc is not available for your operating system " 

I want to know that is why OPENCM9.04 dont work with ROS and linux??? and also about opencm-gcc is not available?

Also when I am using my PC x86_64, i first use arduino to ping the dynamixel, during uploading it says "Enter Bootloader" then after some seconds it says done uploaing, application success. What do it mean "Enter Bootloader"? Does it entering the bootloader or ask from us. I have never faced this issue in windows.

jawad-tech, 2019-07-15:
Reply here please

OpusK, 2019-07-15:
Hi, @jawad-tech 

>  Q2, Q3, 
>  I then used Arduino IDE to code OpenCm9.04 but when I port to arduino and installing OPENCM9.04 library it gives "opencm-gcc is not available for your operating system "

I think it seems to be an architecture (32bit, 64bit) problem.
Did you install the dependency files as described below?
 - [OpenCM9.04 install on linux](http://emanual.robotis.com/docs/en/parts/controller/opencm904/#install-on-linux)

> Also when I am using my PC x86_64, i first use arduino to ping the dynamixel, during uploading it says "Enter Bootloader" then after some seconds it says done uploaing, application success. What do [...]

It is correct that the above log also appears in WINDOWS.
And this means "OpenCM 9.04 has entered the boot loader". I think this expression can cause misunderstanding. However, the person who created the boot loader indicated it, and changing the boot loader can cause many problems, which is difficult to fix.

jawad-tech, 2019-07-15:
Q) Can you answer this please?

I am using Linux ubuntu 16.04 (x86_64) to control the dynamixel servo controllers with Opencm board. I have successfully control the dynamixels with windows using arduino IDE but in linux i am facing some issues:
1.1) When i build the dynamixel-sdk C++ and python libraries to ping the servo, it says TxxRxxerror: No status packet
SO I have followed the issues on github.com dynamixel repository, and do not find the solution.

a) I have tried to reduce the value of timer from 16 msec to 8,4,2,1 msec but all in vain.
b) I am also facing problems, as I followed the tutorial from emanual robotis, when i download 99-dynamixel rule file and place in it the /etc/udev/rules.d but when i check the 
cat /sys/bus/usb-serial/devices/ttyUSB0/latency_timer 
there is no directory. (I have ACM0 instead of USB so I changed the Kernel=/dev/ttyACM* instead of Kernel=/dev/ttyUSB*).

I have downloaded and follow all the tutorials to build by using ROS and linux but when I ping the dynamixels so it dont find any dynamixels. I have also tried position control, but it gives error TxxRxx: The status packet is missing or some error about status packet

OpusK, 2019-07-16:
@jawad-tech 

> but when i check the
cat /sys/bus/usb-serial/devices/ttyUSB0/latency_timer
there is no directory. (I have ACM0 instead of USB so I changed the Kernel=/dev/ttyACM* instead of Kernel=/dev/ttyUSB*).

ACMx does not have a latency_timer file. Therefore, the above setting is meaningless.

Please check the permissions of the port (eg dev / ttyACM0) connected to OpenCM9.04.

jawad-tech, 2019-07-17:
Yes i also tried 
sudo chmod a+rw /dev/ttyACM0
and 
sudo chmod 777 /dev/ttyACM0

The port is ok. 

Still the same error. 
What is the issue then?  Please Read my post and  give  answer to every question please. 

Thanks

What is the possible cause of No status packet RX timeout?

jawad-tech, 2019-07-17:
i also did
modprobe -r ftdi_sio
modprobe ftdi_sio vendor = my usb vendor Id 
product=my product id

When i check the port after it says 
dialout 166 /tty/ACM0 it says like this

OpusK, 2019-07-17:
The current problem is that the opencm-gcc problem and the port are not working properly.

> PC2 with aarch64 architecture: (Using Linux)

Please let me know the exact specifications for the above environment.
The `opencm-gcc` error does not occur in the` linux 64bit` environment I'm testing. In other words, I can not reproduce your situation.

> What is the possible cause of No status packet RX timeout?

There is a possibility that data is not coming into the port mainly or data is lost/collided.
For more information about the SDK, please view the SDK manual or leave an issue in the SDK repository.

> I want to know that is why OPENCM9.04 dont work with ROS and linux???

Please run the ROS example in OpenCM 9.04. OpenCM 9.04 also works in ROS and Linux environments. (Only for small resources)

And, please let me know the connection configuration between OpenCM 9.04 and PC.

I thought that you would not have a problem if you succeeded on Windows10, but please let me know.

And please let me know what code you are using in OpenCM 9.04. (Note that code that is not a basic example is hard to look at)

jawad-tech, 2019-07-17:
I am using the Dynamixel SDK ROS package as found in DYnamixel SDK package. I tried 

rosrun dynamixel_workbench_toolbox find_dynamixel /dev/ttyACM0
It does not find any  dynamixel

OpusK, 2019-07-18:
I have found [the answer](https://github.com/ROBOTIS-GIT/DynamixelSDK/issues/314) to what I asked you here. 
Your usage seems to be wrong. 
We have left the answer in the issue below, so please check it out.
- https://github.com/ROBOTIS-GIT/DynamixelSDK/issues/314

jawad-tech, 2019-07-18:
I understand your point and reached to the solution, but still i have to ask one question that 

in my PC with x86_64 
i executed 
sudo apt-get install libncurses5-dev:i386 it Ran successfully and arduino on linux works cheers  

but on my other PC with aarch64 architecture the libncurses5-dev exists but here you see Its the matter of cross compiling because my aarch64 IS different from amd i386 so i need a library for cross compiling.
Suggest me please

jawad-tech, 2019-07-18:
the libncurses5-dev:i386 doesn't work on my PC with  aarch64

OpusK, 2019-07-18:
@jawad-tech 

The current OpenCM 9.04 compiler uses [GNU-ARM-Embedded](https://launchpad.net/gcc-arm-embedded/+series).

And the current version is [`5-2016-q2-update`](https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q2-update).

There is a release note and README in the link above, which shows support for Arm architecture 64bit.

The README contains the following precautions.
> For 64 bit system, 32 bit libc and libncurses are required to run the tools.

Currently I do not have a PC to test aarch64, so it is difficult to know which package to install and whether it works.
This is due to the dependencies of the operating system on the system, so I need to find out about it.

jawad-tech, 2019-07-18:
can you please try it and provide me a solution.

OpusK, 2019-07-18:
@jawad-tech 

> can you please try it and provide me a solution.

Unfortunately, there is no environment that I can test. However, there is a suggestion that I made to the previous content. Please come to GitHub and check the issue. (Not mail)

jawad-tech, 2019-07-18:
actually i am a beginner in linux so need some expert opinion please. Kindly help me. I will also try with my rookie skills

jawad-tech, 2019-07-18:
Well itried to build the toolchain but it still stays that tolls_opencm are not present for your architecture, when i installed opencm from boards manager in arduino IDE.

the steps i followed to install the GNU-ARM-EMBEDDED are 

1) sudo add-apt-repository ppa:team-gcc-arm-embedded/ppa
2) sudo apt-get update
3) sudo apt-get install gcc-arm-embedded

OpusK, 2019-07-19:
@jawad-tech 

Have you tried the links below that I mentioned above?
The content you wrote is different from the link I gave you.

> +Add: I can not test it myself, but I found the link below.
[Running 32 bit applications on Aarch64 By James Kingdon, July 22, 2017](https://forum.armbian.com/topic/4764-running-32-bit-applications-on-aarch64/)

closed 2019-07-30 by OpusK