ROBOTIS-GIT/OpenCM9.04 #88: Trying to run the I2C in slave mode

YifanAtBioRoboLab, 2019-08-23:
Hi, I'm trying to connect the openCM9.04 board to some other control unit and thinking about to receive command through hardware I2C. Is there any possible way to let the openCM side be in slave mode, receive command and send back data? It will be better if this can be done through DMA or interruptions

KyoungHwan2046, 2019-09-16:
Hi, @YifanAtBioRoboLab 
I'm sorry for reply late. 
I will check this issue. 
Thank you.

pranavb104, 2019-10-15:
Im also trying to get the opencm run in slave mode...is there anything available yet?

KyoungHwan2046, 2019-10-16:
Hi, @YifanAtBioRoboLab 

We are getting ready for update of slave mode examples.

I'm sorry for update late.

KyoungHwan2046, 2019-10-24:
Hi @YifanAtBioRoboLab 

The update will be in next year.
Some libraries of opencm and opencr are not latest version.
You can run the i2c in slave mode with latest wire library.
Please read this web site.
>https://circuitdigest.com/microcontroller-projects/stm32-stm32f103c8-i2c-communication-tutorial

And you can update the wire library in
>https://github.com/stm32duino/Arduino_Core_STM32/tree/master/libraries/Wire

KyoungHwan2046, 2019-11-05:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-11-05 by KyoungHwan2046

nxdefiant, 2020-11-21:
Looks like the libs have not been updated yet, any news on this issue?

nxdefiant, 2020-11-21:
I have managed to get basic I2C reception working using the HAL api:
```
#define BUF_LEN 8

int led_pin = 14;
int i2c_status;
I2C_HandleTypeDef hi2c;
uint8_t rxbuffer[BUF_LEN];
uint32_t loop_run=0;

void setup() {
	// Set up the built-in LED pin as an output:
	pinMode(led_pin, OUTPUT);
	digitalWrite(led_pin, HIGH); // Off

	Serial.begin(115200);
	memset(&hi2c, 0, sizeof(I2C_HandleTypeDef));
	hi2c.Instance = I2C2;
	hi2c.Mode = HAL_I2C_MODE_SLAVE;
	hi2c.Init.OwnAddress1 = 0x60;
	hi2c.Init.AddressingMode = I2C_ADDRESSINGMODE_7BIT;
	i2c_status = HAL_I2C_Init(&hi2c);
	if (i2c_status == HAL_OK) {
		digitalWrite(led_pin, LOW);
 	}
}

void loop() {
	memset(rxbuffer, 0, BUF_LEN);
	if (HAL_I2C_Slave_Receive(&hi2c, (uint8_t *)rxbuffer, 5, HAL_MAX_DELAY) == HAL_OK) {
		Serial.println(rxbuffer[0]);
		Serial.println(rxbuffer[1]);
		Serial.println(rxbuffer[2]);
		Serial.println(rxbuffer[3]);
		Serial.println(rxbuffer[4]);
	}

	loop_run++;
	digitalWrite(led_pin, loop_run%2 == 0);
}
```