ROBOTIS-GIT/OpenCM9.04 #90: Serial Port problems with <ros.h> on arduino IDE

domiinio, 2019-09-23:
Hello everyone,

I'm trying to control a dynamixel ax-12a with the OpenCM9.04 board and I'm programming the board using the arduino IDE.

I was using the Dynamixel Workbench library for commanding the servo and it was working just fine, but because I will need to use ROS in my project, I wanted to include it into my code already.

I downloaded the rosserial library and included it into simple h_Position example (under the OpenCM9.04/Dynamixel Workbench examples ) like:

...
#include <DynamixelWorkbench.h>
#include <ros.h>
...

but as soon as I included the <ros.h> I started getting the following error
In file included from /home/dominik/Arduino/libraries/Rosserial_Arduino_Library/src/ros.h:39:0,
from /tmp/arduino_modified_sketch_218835/h_Position.ino:20:
/home/dominik/Arduino/libraries/Rosserial_Arduino_Library/src/ArduinoHardware.h: In constructor 'ArduinoHardware::ArduinoHardware()':
/home/dominik/Arduino/libraries/Rosserial_Arduino_Library/src/ArduinoHardware.h:55:16: error: cannot convert 'USBSerial*' to 'HardwareSerial*' in assignment
iostream = &Serial;

and the code won't compile.

Does anyone know, what is causing this error and how to work around it?

Thanks in advance!

KyoungHwan2046, 2019-09-24:
Hi, @domiinio 
I just checked with h_position.ino file added '#include <ros.h>' line, and the compile and upload was good.

So, I want to know more detail about your code.
Do you mean that you just only added the '#include <ros.h>' line and the code won't compile??

Please let me know more information about your situations. 

Thanks you.

domiinio, 2019-09-24:
Hi @KyoungHwan2046 
the problem I'm having occurs only on Ubuntu (I'm running the 18.04 LTS version). When I try the same code on Windows, it compiles just fine, so I'm guessing it has to do something with the linux - ros.h interactions or the ros.h implementation for Ubuntu, since I encountered this error on multiple PC's, that are running Ubuntu 18.04.

To your question, yes, just this one include '#include <ros.h>' is enough to cause the error. I tried compiling just the h_Position without any extra lines of code or includes and it compiled and uploaded fine, but as soon as I added the <ros.h> library, I got the error again.

I'm sharing the whole error message, in case it helps.
------------------------------------------------------------------------------------------------------------------------------
In file included from /home/dominik/Arduino/libraries/ros_lib/ros.h:43:0,
                 from /tmp/arduino_modified_sketch_714154/h_Position.ino:20:
/home/dominik/Arduino/libraries/ros_lib/ArduinoHardware.h: In constructor 'ArduinoHardware::ArduinoHardware()':
/home/dominik/Arduino/libraries/ros_lib/ArduinoHardware.h:79:16: error: cannot convert 'USBSerial*' to 'HardwareSerial*' in assignment
       iostream = &Serial;
                ^
Multiple libraries were found for "DynamixelWorkbench.h"
 Used: /home/dominik/.arduino15/packages/OpenCM904/hardware/OpenCM904/1.4.2/libraries/DynamixelWorkbench
Multiple libraries were found for "DynamixelSDK.h"
 Used: /home/dominik/.arduino15/packages/OpenCM904/hardware/OpenCM904/1.4.2/libraries/DynamixelSDK
Multiple libraries were found for "ros.h"
 Used: /home/dominik/Arduino/libraries/ros_lib
 Not used: /home/dominik/.arduino15/packages/OpenCM904/hardware/OpenCM904/1.4.2/libraries/ros_lib
 Not used: /home/dominik/Arduino/libraries/Rosserial_Arduino_Library
exit status 1
Error compiling for board OpenCM9.04 Board.

------------------------------------------------------------------------------------------------------------------------------

I also tried changing the 'iostream' to USBSerial1 in ArduinoHardware.h but it didn't seem to help. 
But it can very well be, that I changed it in the wrong place or it should have been something else replacing it.

Thanks for the reply

KurtE, 2019-09-24:
As I mentioned up on Trossen forum, I think the RosSerial code needs to change to have a section in it that is very much like the one that is in there for Teensy boards (PJRC).

https://github.com/ros-drivers/rosserial/blob/melodic-devel/rosserial_arduino/src/ros_lib/ArduinoHardware.h#L44

Actually it may depend on which version of this library you have.  I see at least a few different versions. 
But it you look up at maybe: 
I think the github version of the Arduino install is at: 
https://github.com/frankjoshua/rosserial_arduino_lib/blob/master/src/ArduinoIncludes.h#L44

I think his released version is several mods old. 

But if you look there and maybe add a section like:
```
#elif defined(__OPENCM904__) 
  #include <USBSerial.h>  // Teensy 3.0 and 3.1
  #define SERIAL_CLASS USBSerial
```

You might see if that works.  Note: This definition location may change depending on which version you have...

KyoungHwan2046, 2019-11-20:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-11-20 by KyoungHwan2046

ProfJust, 2022-11-18:
This error occures if there are several ros_serial / ros_lib - Libraries using <ros.h> in the Arduino/Libraries-Folder.