ROBOTIS-GIT/OpenCM9.04 #92: Issues about sync read and its examples

YifanAtBioRoboLab, 2019-10-29:
While attempting to apply the sync read method to get actuator data, there are several issues coming out.

First, we noticed when their's no actuator connected to the controller (openCM9.04 for our case), The controller will crash instead of printing out error messages. This issue stands true even to the example code (m_Sync_Read_Write). We wish someone can look into this issue so we don't need to worry about the circumstance when the connection is broken for any weird reason.

Another issue. We were planning to control 12 actuators using 1 single controller. But we noticed when the total number exceeds 10,  the sync read method will response an error message "[TxRxResult] There is no status packet!". Is there any way to make the sync read work with more than 10 actuators?

KyoungHwan2046, 2019-10-31:
Hi, @YifanAtBioRoboLab 

We will check this issue and work on it soon.

Thank you.

JaehyunShim, 2020-10-16:
Please leave your question at this [link](https://github.com/ROBOTIS-GIT/DynamixelSDK/issues) If your question is about the usage of Dynamixel-SDK,

closed 2020-10-16 by JaehyunShim