ROBOTIS-GIT/OpenCM9.04 #93: Error in example "n_Bulk_read_write" (groupBulkWrite addparam failed)

GrigoriyAY, 2019-12-23:
I plan to use code from example "n_Bulk_read_write" on OpenCM 9.04 with Expansion board 485, but I got following messages (Windows, Arduino IDE, library: DynamixelWorkbench):

```
Succeeded to init : 1000000
Succeeded to ping
id : 1 model_number : 29
Succeed to change joint mode
Succeeded to ping
id : 5 model_number : 320
Succeed to change joint mode
[DynamixelDriver] Succeeded to init groupBulkWrite!
[DynamixelDriver] Succeeded to init groupBulkRead!
[DynamixelDriver] Succeeded to add param for bulk read!
[DynamixelDriver] Succeeded to add param for bulk read!
groupBulkWrite addparam failed
Failed to add bulk write led param
groupBulkWrite addparam failed
Failed to add bulk write position param

Failed to bulk write
```
Could you help, what can be wrong?
Example k_Read_Write works good, but I need to control 4 pairs of Dynamixel motors (four MX 106T and four MX28T) simultaneously. Using the independent data transfer of the desired position in cycle "for" I have very slow movement of the servomotors.

KyoungHwan2046, 2019-12-24:
Hi, @GrigoriyAY 

I've check the code with five dynamixel, and it does works well. 

groupBulkWrite addparam failed <- this error code means that wrong ID used.

please check the param of dxl_id[0] at the
 "result = dxl_wb.addBulkWriteParam(dxl_id[0], "Goal_Position", goal_position[0], &log); "
line.

Thank you.

GrigoriyAY, 2019-12-24:
Hello, @KyoungHwan2046 
As you can see in my previous message ID of servomotors were successfully defined. And I changed only numbers of IDs and following lines:
```
#define BAUDRATE  1000000
int32_t goal_position[2] = {2200, 2048};
Serial.begin(115200);
```

KyoungHwan2046, 2020-02-04:
Hi , @GrigoriyAY 
I'm sorry for the late reply.
Do you still have this problem?
On your question, the ID are 1 and 5 at first time, 
but on your answer, the following message said the ID are 2 and 6.
which IDs are right?

GrigoriyAY, 2020-02-04:
@KyoungHwan2046, hello

I just try to control 8 servomotors simultaneously, it was another pair of motors in second message. Yes, the problem is still actual.

GrigoriyAY, 2020-03-11:
@KyoungHwan2046 yes, but I use it with MATLAB.

JaehyunShim, 2020-10-16:
I will close this thread as there have been no recent updates.

closed 2020-10-16 by JaehyunShim