ROBOTIS-GIT/OpenCM9.04 #94: [OpenCM904 Not Detected UART Flash Loader]

Ash-S, 2020-02-18:
Hi, I am trying to set up OpenCM904 with Serial1 (TX connected to USB-RS232 dongle to access it through COM port on "Flash Loader Demonstrator" from ST Microelectronics, since it has STM32F103CB MCU. Once i connect and access through com port, it says:

[image]

Please let me know how can i reset it to get detect on com port or suggest else to solve this.
Thanks!

ROBOTIS-Will, 2020-02-24:
@Ash-S 
Hi, OpenCM9.04 should not be flashed with other software than Arduino IDE or Robotis software as it could erase the bootloader section of memory. Since your inquiry will void the product warranty, please understand that we cannot disclose the instruction on flashing OpenCM9.04 with 3rd party tools.
Thank you.

Ash-S, 2020-02-24:
@ROBOTIS-Will Hi, thanks for your reply. I understand it but I didn't mean to flash bootloader or so, I mean, if i wish to upload my application (sketch) compiled binary, how can i do that except using Arduino/openCM IDE. I don't want to change bootloader, just need to upload *.bin to openCM without using or installing IDE.
Hope you understand my point. Please suggest!

ROBOTIS-Will, 2020-03-06:
@Ash-S 
Oh, I see.
If you want to upload the `.bin` file generated from Arduino IDE, you can use `opencm9.04_ld.exe` if you are running on Windows. It is included in the `Arduino 15 > packages > OpenCM904 > tools > opencm_tools > 0.0.2 > win`.
You can run as below to upload the binary into OpenCM9.04, but you should modify the com port and baudrate properly.
```bash
OpenCM904ld.exe COM1 57600 <filename.bin> 1  opencm
```

closed 2020-08-28 by ROBOTIS-Will