ROBOTIS-GIT/OpenCM9.04 #98: openCM904C Serial Port Not working

Ash-S, 2020-12-01:
I am interfacing TOF sensor on openCM904C (via Serial3 UART/Serial2 UART). Seems, its not detecting `Serial3.available()`. even I tried tested the example "d_Serial3_Echo" but no response. Please let me know if I am missing something here. Please suggest.

ROBOTIS-Will, 2020-12-07:
Hi,
The **Serial2_Echo** or **Serial3_Echo** examples read and write to its own serial port.
In order to read the data of the Serial3 via other serial port, you have to relay the data to the desired serial.
When reporting the issue, please describe in detail about what you've done for hardware connection and software programming, or else there's not much thing I can do with extremely limited information about your environment.
Thank you.

Ash-S, 2020-12-07:
Hi @ROBOTIS-Will , thanks for your answer but since first I am simply trying to test the provided example (on openCM904) as explained below, nothing happens. This is what defined in example code (d_Serial3_Echo). Here are the details, what have done so far with hardware/software. 

```
Serial1 : Dynamixel_Poart
 Serial2 : Serial_Poart(4pin_Molex)
 Serial3 : Serial_Poart(pin26:Tx3, pin27:Rx3)

 TxD3(Cm9_Pin26) <--(Connect)--> RxD(PC)
 RxD3(Cm9_Pin27) <--(Connect)--> TxD(PC)
 */

void setup(){
  //Serial3 Serial initialize
  Serial3.begin(57600);
}
void loop(){
  // when you typed any character in terminal
  if(Serial3.available()){
    //print it out though USART2(RX2,TX2)
    Serial3.print((char)Serial3.read());
  }
}
```
I didn't see any of D26, D27 pins on OpenCM904 as defined in 'd_Serial3_Echo' code to connect it to PC. Seems it should be D24, D25. Is that correct?

from [https://emanual.robotis.com/docs/en/parts/controller/opencm904/]
Serial/SerialUSB | USB Port
Serial1: D11, D12; DXL Port
Serial2: A4, A5   
Serial3: D24, D25

To read the send/receive data from same 'Serial3', I connected the hardware like this....
-connect 'usb to ttl' dongle to PC (usb) and to openCM904 'Serial3' port (Tx-->Rx, Rx-->Tx, Gnd-->Gnd)
-since i already uploaded the compiled code "d_Serial3_Echo" to openCM through serialUSB. I simply run it and check it on serial monitor (USBtoTTL PC port COM5) on openCM. Nothing happens when send anything. Please suggest.

Thanks!

ROBOTIS-Will, 2020-12-08:
Hi @Ash-S 
Thank you for the updates.
You are correct for the Serial pins. It looks like the code has wrong information about the pin number.
Disregard the information and use the information written in the eManual.
Serial3_Echo should return your terminal input and display it on the same terminal.
I've just tested it and seems it is working fine.
[image]

Ash-S, 2020-12-08:
Hi @ROBOTIS-Will, thanks for your suggestion. Since I don't have LN-101, I tested it with similar kind of `"USB to TTL Dongle"` connected at 24(Tx) and 25(Rx) and GND of OpenCM904 instead. I tested run the same code `"d_Serial3_Echo",` but I didn't get anything on terminal. Thereafter I tried to test to send the data from `"SerialUSB to Serial3"`, it works but opposite way doesn't work `(from Serial3 to SerialUSB)` and found that there is issue with detecting Serial3 at ``"if (Serial3.available)"`` .

Could you please share some more links or example you tested? I didn't understand what is causing the issue. Please!

ROBOTIS-Will, 2020-12-08:
I modified the `c_serial_Serial2_Converter` into serial-serial3 as below.
```
#define BAUDRATE 57600

#ifndef BAUDRATE
uint32_t g_usb_baudrate;
#endif

void setup(){
  while (!Serial && millis() < 5000) ;
  Serial.begin(115200);
#if defined(BAUDRATE)
  Serial3.begin(BAUDRATE);
#else
  g_usb_baudrate = Serial.getBaudRate();
  Serial3.begin(g_usb_baudrate);
#endif  
  pinMode(BOARD_LED_PIN, OUTPUT);
}

void loop(){
  if(Serial.available()){
    Serial3.print((char)Serial.read());//send data coming from USB to Serial3
  }

  if(Serial3.available()){
    toggleLED();
    Serial.print((char)Serial3.read()); //send data coming from Serial3 to USB(PC)
  }
#ifndef BAUDRATE
  if (g_usb_baudrate != Serial.getBaudRate()){
    g_usb_baudrate = Serial.getBaudRate();
    Serial3.begin(g_usb_baudrate);
  }
#endif  
}
```
**Connection Description**
* `Pin 24(TXD_3) of OpenCM9.04` to `RX pin of USB-Serial device`
* `Pin 25(RXD_3) of OpenCM9.04` to `TX pin of USB-Serial device`
* `USB-Serial device` is connected to a PC, `COM9` port.
* `OpenCM9.04` is connected to a PC via USB cable, `COM7` port.

[image]

Ash-S, 2020-12-09:
Hi @ROBOTIS-Will , thanks for sharing the test code. well, I followed the same as mentioned and I can see the data sent from `openCM to PC` but unfortunately I am unable to see the data sent from `PC to openCM904`. this is the same what i was facing earlier (`SerialUSB to Serail3` works fine but `Serial3 to SerialUSB` doesn't). here is the log.....please refer.

[image]

what I am missing here? serial driver or firmware something? please suggest. thanks

Ash-S, 2020-12-09:
Hi @ROBOTIS-Will , thanks for sharing the test code. I followed it and found that....
-sending data from `openCM to PC `works fine.
-sending data from `PC to openCM `doesnt work.

So, I tested it separately (mean only openCM904) with serial connection (`pin 24,25`) and it works well. But since earlier I was using it with openCM 485 EXP, It is showing the issue while sending data from PC to openCM. May I please know why it doesn't work when plugged into EXP 485 but works well individually (without EXP485).

[image]

since i want to use it with EXP485 because I have connected dynamixels to EXP board and I wish to use openCM serial 3 for sensor connections. Please suggest. Thanks

ROBOTIS-Will, 2021-01-25:
OpenCM9.04 board uses a direction control pin for Serial3 that connects to OpenCM485EXP since DYNAMIXEL is hald duplex asynchronous communication.
In order to read from or write to DYNAMIXEL via Serial3, you should also control the DXL_DIR3 pin which is 22 pin of OpenCM9.04.
Set this pin to HIGH when transmitting packet to DYNAMIXEL, and set to LOW when receiving packet from DYNAMIXEL.
Thank you.

Ash-S, 2021-02-03:
Thank you @ROBOTIS-Will for kind explanation. :)