ROBOTIS-GIT/OpenCM9.04 #99: Can't connect to MCU with ST-LINK/V2 and ST-LINK Utility

LeTo37, 2020-12-08:
Hi there,

I'm sure I am doing something wrong but I would really appreciate a bit of help on this. 

I am simply trying to connect to the OpenCM9 with ST-Link Utility. My hardware connection is as advised on the Robotis website:
[image]

I have separately powered the CM9 with 5V.

No matter what target settings I change, I consistently cannot connect to the board:
[image]

Any ideas?

Thank you in advance!

ROBOTIS-Will, 2020-12-09:
Hi @LeTo37 
I just tried with my OpenCM9.04 and ST-Link/V2 based on the connection from the eManual, but couldn't find the problem(memory is protected by default)
[image]
Please be aware when you program the device with ST-Link, it will erase the bootloader section which cannot be recovered as well as void the warranty.
Thank you.

LeTo37, 2020-12-09:
Hi Will,

Thank you so much for your quick response! I really appreciate it, especially given the ST-Link has nothing to do with ROBOTIS.

I managed to solve my issue with a slightly different connection to that shown in the eManual (shown below). By holding the reset button down while connecting and choosing "Connect Under Reset" in my Target settings, I managed to establish a connection.
[image]

closed 2020-12-09 by LeTo37

ROBOTIS-Will, 2020-12-10:
Oh, thank you for the updates!
We'll release a new product very soon that works with Arduino MKR(which is as small as OpenCM9.04), and you might also want to [check it out](https://emanual.robotis.com/docs/en/parts/interface/mkr_shield/).