ROBOTIS-GIT/OpenCM9.04 #101: br410 uart over ble profile code

olsky, 2021-01-22:
Hello Team,

I have a question on the bt dongle, which we use to connecto to the bt410 receiver, it has UART over BLE profile. And of course all is working.

Now we wanted to integrate all with our custom nrf52 board. So we would need the UART profile in order to connect to the receiver and send AT commands (to control motors and so on).

Could you please share some details or point to the Profile code?

Thank you in advance.

ROBOTIS-Will, 2021-01-25:
Hi @olsky 
Thanks for your inquiry. I'll get back to you as soon as I hear back from the development team.

irarykim, 2021-01-25:
Hi @olsky 
BT-410 Slave Module, BT-410 Master Module and BT-410 Dongle all uses Nordic UART Service.
Please refer following link: https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.sdk5.v14.0.0%2Fble_sdk_app_nus_eval.html

However, BT-410 Master Module and BT-410 Dongle can ONLY pair with a NUS Peripheral device with MAC address DBE7DFxxxxxx.

Thank you