ROBOTIS-GIT/OpenCM9.04 #102: Are they a set? OpenCM9.04 Release 1.5.0 and OpenCM9.04 Firmware ver 23.

ROBOTIS-Shibata, 2021-02-01:
Is it necessary to update OpenCM9.04 to FW version 23, using R+Manager 2.0, in order to we use OpenCM9.04 Release 1.5.0?
[image]

ROBOTIS-Will, 2021-02-01:
Hi @ROBOTIS-Shibata 
The OpenCM9.04 firmware(v23) in R+Manager is an official firmware for ROBOTIS MINI and ROBOTIS software.
On the other hand, the board manager(1.5.0) from this GitHub repository is a set of library to help developers programming OpenCM9.04 in Arduino IDE.
Please note that when programming with Arduino IDE, the OpenCM9.04 does not work with ROBOTIS softwares until recovering the firmware with R+Manager.
Thanks.

ROBOTIS-Shibata, 2021-02-02:
Since the update times of both sides overlapped, I wanted to check if it was related to anything.
Thank you. It's a close.

closed 2021-02-02 by ROBOTIS-Shibata