ROBOTIS-GIT/OpenCM9.04 #103: OpenCM 9.04 model for circuit design programs

Y-Grigoriy, 2021-02-13:
Hello,

I saw pictures of the OpenCM connection with different devices in manual (e.g. https://emanual.robotis.com/assets/images/parts/controller/opencm904/sdcard_bb.png). It looks like scheme drawn in program for electrical circuit drawing (e.g. Fritzing). Do you have the file with OpenCM model (may be .fzpz for Fritzing)? If yes, could you share it?

ROBOTIS-Will, 2021-02-16:
Hi @Y-Grigoriy 
I've found the Fritzing file, but it didn't contain any circuitry information.
In case you need schematics and other information about the board, please refer to the OpenCM9.04 eManual below.
https://emanual.robotis.com/docs/en/parts/controller/opencm904/#downloads
Thank you.

Y-Grigoriy, 2021-02-16:
Hi @ROBOTIS-Will 
If this file contains good image of OpenCM in svg format, it also be helpful. I can supplement this file by necessary schemes by myself. What about Dynamixel servomotors as parts for Fritzing?

ROBOTIS-Will, 2021-02-17:
Let me see if I can find and share any meaningful data from what I found.
Unfortunately, I don't see any DYNAMIXEL in this project.

Y-Grigoriy, 2021-03-31:
@ROBOTIS-Will Do I right understand, that you couldn't find any data?

ROBOTIS-Will, 2021-04-01:
@Y-Grigoriy 
I only found .svg files for OpenCM9.04 and XL-320.
I'll upload them here just in case you need them.
[robotis_fritzing.zip](https://github.com/ROBOTIS-GIT/OpenCM9.04/files/6240218/robotis_fritzing.zip)
Thanks!

Y-Grigoriy, 2021-04-02:
@ROBOTIS-Will thank you!

closed 2021-04-02 by Y-Grigoriy