ROBOTIS-GIT/OpenCM9.04 #104: Will Wire.h be implemented in hardware?

brianjohnzhang, 2021-02-27:
I am trying to use a variety of I2C sensors in a relatively simple way. However, many of the sensors I want to use only have libraries that rely on Wire.h. While it is possible for me to adapt some of the libraries to use I2Cdev, I would really prefer to avoid it since my understanding of I2Cdev and Wire are both pretty low and my priority is getting the robot to work. 

I've read into the prior issues (#91. #88, #73, #52, #14) and I've seen that there was a brief discussion/pointer towards the Arduino_Core_STM32 libraries (https://github.com/stm32duino/Arduino_Core_STM32/tree/master/libraries/Wire). I gave it a brief attempt but the amount of understanding/integration for the background libraries is beyond me. 

It looked like there may have been plans to integrate this library (https://github.com/ROBOTIS-GIT/OpenCM9.04/issues/88#issuecomment-545739062). Will this occur anytime soon? Otherwise, will the Dynamixel Shield for Arduino MKR  (https://www.robotis.us/dynamixel-shield-for-arduino-mkr-series/) be released soon?

ROBOTIS-Will, 2021-03-02:
Hi @brianjohnzhang 
Thank you for your interest in OpenCM9.04.
Unfortunately, we do not have an assigned schedule on updating the OpenCM9.04 library.
However, DYNAMIXEL Shield for Arduino MKR series will be released very soon.

brianjohnzhang, 2021-03-03:
Okay, thank you for the information. It is unfortunate for the OpenCM board but we will shift towards the MKR shield instead. 

Due to the limitations of the boards usable with the shields, it would be nice to see a condensed non-shield format controller that simply uses a serial, SPI, or I2C port, though I'm sure you have more insight into what products people may want.

closed 2021-03-03 by brianjohnzhang

ROBOTIS-Will, 2021-03-03:
@brianjohnzhang 
Thank you for your understanding.
For the same reason, I also prefer OpenCM9.04 over Arduino + shield combination.
Wish the best for your and please feel free to let us know if you have any feedback on using DYNAMIXEL Shield with Arduino boards.
Thank you.