ROBOTIS-GIT/OpenCM9.04 #106: How to modifying the PWM frequency on the OpenCM9.04 ?

adluthi, 2021-08-16:
Hi all,

I am using the OpenCM9.04 board on Arduino and I want to control the frequency of the PWM generated but I don't find any way to do it....

I have found on the [link](https://emanual.robotis.com/docs/en/software/opencm_ide/api_reference/) at the section  "4.5.5.4. Arduino Compatibility" but I don't find which timer controls the PWM signal (point 1 on the 4.5.5.4 part)...

Does anyone have an idea how to do that please? 

Thank you in advance!

ROBOTIS-Will, 2021-08-17:
Hi @adluthi 

PWM supported GPIO pins and timers are described in the below.
https://github.com/ROBOTIS-GIT/OpenCM9.04/blob/master/arduino/opencm_arduino/opencm9.04_release/variants/OpenCM904/variant.cpp#L38-L51

The PWM driver is implemented as below.
https://github.com/ROBOTIS-GIT/OpenCM9.04/blob/master/arduino/opencm_arduino/opencm9.04/variants/OpenCM904/hw/driver/drv_pwm.c

Thank you.

adluthi, 2021-08-17:
Hi @ROBOTIS-Will, 

Thank you for your answer however since I am a new user with the OpenCM board and also because I am coding in Arduino, I don't know how the files you sent me could help me... 

For example, I have done this code which allows me to set a duty cycle through serial communication but I really don't get how to tune the frequency. The setPeriod() function does not change anything. Can you guide me, please?

Thank you,

`
int pwm_pins = 12;
const byte numChars = 32;
char receivedChars[numChars];   // an array to store the received data
boolean newData = false;

HardwareTimer timer(TIMER_CH1);

void setup() {
  Serial.begin(9600);
  timer.setPeriod(5000);
}

void loop() {
  // put your main code here, to run repeatedly:
  uint8_t pwm_out = 0;

  recvWithEndMarker();
  setPWM();
}
  
 

void recvWithEndMarker() {
    static byte ndx = 0;
    char endMarker = '\n';
    char rc;
    
    while (Serial.available() > 0 && newData == false) {
        rc = Serial.read();

        if (rc != endMarker) {
            receivedChars[ndx] = rc;
            ndx++;
            if (ndx >= numChars) {
                ndx = numChars - 1;
            }
        }
        else {
            receivedChars[ndx] = '\0'; // terminate the string
            ndx = 0;
            newData = true;
        }
    }
}

void setPWM() {
    if (newData == true) {
      analogWrite(pwm_pins, atoi(receivedChars));
      Serial.println(atoi(receivedChars));
      newData = false;
    }
}`