ROBOTIS-GIT/OpenCM9.04 #108: Cant't make OpenCM communicate with sensors via I2C

yuya-0411, 2022-05-27:
Hello, @ROBOTIS-Will 
I'm Yuya.

I use OpenCM9.04 C Type for my study.
I tried using MPU9250 (IMU sensor) with OpenCM via I2C.
I checked I2C with OpenCM [your guide](https://emanual.robotis.com/docs/en/parts/controller/opencm904/).

But I can't make it communicate with sensors via I2C.
First of all, without the below code, somehow I2C device was found at all addresses even though I connect only one I2C device to OpenCM.

```
pinmode(24, INPUT_PULLUP)
pinmode(25, INPUT_PULLUP)
```

[image]

If I add the above code, the result is below.

[image]

I did some tests to find the causes for this problem.

(1)
I suspected sensor failure for the first time. To check that, I did it with Arduino UNO.
I could make Arduino communicate with a sensor via I2C.

[image]

(2)
I suspected  VCC(3.3V) and GND on OpenCM. 
This also succeeded.

From (1), and (2), I guessed this problem is related to OpenCM.

[image]

If you have any solution to this problem, please teach me.
I would like to I2C device with OpenCM9.04 C Type.

thank you for your attention

ROBOTIS-Will, 2022-05-30:
Hi @yuya-0411 

I can't find any information about your code.
I'd recommend to review the MPU6050 example below and start from there.
https://emanual.robotis.com/docs/en/parts/controller/opencm904/#mpu6050-dmp

I'm sorry but I can't provide technical support for debugging custom libraries or source codes.
Thank you.

yuya-0411, 2022-05-30:
@ROBOTIS-Will 
Thank you for your rapid response.

I'm sorry. I forgot to explain the code in detail.
I just ran Arduino example "Who Am I" and "I2C check" in order to check I2C communication because I suspected that the problem was related to not Software but Hardware.

Actually, the I2C device worked on Arduino Uno.
I will try what you suggested.

thank you

closed 2022-05-30 by yuya-0411