ROBOTIS-GIT/OpenCM9.04 #111: Arduino IDE Board Manager cannot install OpenCM board

jatin-shah, 2023-11-15:
Documentation calls for installing OpenCM package using file https://raw.githubusercontent.com/ROBOTIS-GIT/OpenCM9.04/master/arduino/opencm_release/package_opencm9.04_index.json

However, it is unable to install the OpenCM package. 

It appears the package_opencm9.04_index.json file is pointing to Github URLs that are no longer valid! (The github repos are now tagged repos. So the URL has changed?!)

Anyone know how to get around this and install the OpenCM package manually.

shiba-8ro, 2023-11-20:
@jatin-shah 
Hi,
I have successfully installed OpenCM9.04 using that URL, Arduino IDE is 1.8.12.
You may have problems with the newer Arduino IDE.

I am afraid this is not the purpose of your question, but I am recommending you to switch to OpenRB-150.
The reason is that OpenCM9.04 will be discontinued in the near future.
At this time, OpenCM9.04 is Out of stock. It will probably not be available as is.
https://en.robotis.com/shop_en/item.php?it_id=902-0084-040

TuxLeon, 2023-11-20:
For me the give link to the JSON file works fine in the new Arduino IDE (2.2.1) under LINUX
The same JSON doesn't work in the MAC Versions of the Arduino IDE.