ROBOTIS-GIT/OpenCM9.04 #112: Arduino IDE - Uploading error: exit status 0xc0000135

Rouno, 2024-08-02:
Hi guys, just reposting a problem [I published on Arduino forum](https://forum.arduino.cc/t/uploading-error-exit-status-0xc0000135/1287463)

I've just changed my laptop from windows 10 to windows 11, Arduino IDE 2.3.2, OpenCM board package version 1.5.1
I had never had any problem until now : I can compile any program but I cannot upload the binary to the OpenCM

`Sketch uses 33840 bytes (28%) of program storage space. Maximum is 116736 bytes.
Global variables use 8640 bytes of dynamic memory.
"C:\Users\rdidier\AppData\Local\Arduino15\packages\OpenCM904\tools\opencm_tools\0.0.2/win/opencm9.04_ld.exe" "COM6" "57600" "C:\Users\rdidier\AppData\Local\Temp\arduino\sketches\2DD9910FD5BBE52B25AC84BD77F90A83/a_Bare_Minimum.ino.bin" "1" "opencm"
Failed uploading: uploading error: exit status 0xc0000135`

The serial monitor works though, I can totally send and receive messages through the virtual COM port

Can you please help ?

Rouno, 2024-08-02:
Hi guys, just to let you know : I switched back to Arduino 1.8 IDE and I don't experience the problem anymore.

I will test against 2.x later on to see if the situation will have improved

shiba-8ro, 2024-08-02:
Although it is not a direct solution to this issue, since OpenCM9.04 was discontinued, one solution is to replace to OpenRB-150.